arduino/Arduino #3026: Cannot reuse SD Card library after card removed and inserted.

squelched, 2015-04-23:
A fix might come down to a 3 byte change in sd.cpp, but I am unsure whether the author's code was deliberate or accidental.

A global SDClass instance SD is defined in sd.cpp. In SDClass.open(), the code "SD.root" refers to this instance. Instantiating another instance of SDClass will continue to refer to the global instance. I believe this to be a bug. The correction would be "root". This would enable any instances of SDClass to refer to their own private class member, not the global member. A simple 3 byte change.

I found this bug because if I use the SD object to initialize an SD card with SDClass::begin(), then used a File object linked to the global object to open a file, then closed the file, if the SD card was then removed and inserted, File::open() would fail and even SDClass::begin() would fail. I don't think begin() was meant to be called twice. So I just avoided the use of the global SD object and instantiated an SDClass object on the stack. That's when I discovered that it too would fail and traced that to the reference to the global SDClass object, SD. The 3 byte change fixed that problem. Now I can remove and replace an SD Card.

facchinm, 2015-04-24:
Hi @squelched , can you submit a PR with your modifications? 
The idea seems good however :wink:

squelched, 2015-04-26:
Okay. Done. I think. I'm a GitHub newbie. Cheers

facchinm, 2015-04-27:
Fixed with 1c25d24

closed 2015-04-27 by facchinm