arduino/Arduino #3028: 1.6.4 avr-g++: error: {compiler.warning_flags}: No such file or directory

mschlenker, 2015-04-24:
This happened after updating the toolchain to 1.6.4. Ubuntu 14.10, 64 bit:
 
```
/home/mattias/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++ -c -g -Os {compiler.warning_flags} -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/mattias/.arduino15/packages/arduino/hardware/avr/1.6.4/cores/arduino -I/home/mattias/.arduino15/packages/arduino/hardware/avr/1.6.4/variants/standard /tmp/build3147555964145188522.tmp/sketch_apr24c.cpp -o /tmp/build3147555964145188522.tmp/sketch_apr24c.cpp.o 
avr-g++: error: {compiler.warning_flags}: No such file or directory
Fehler beim Kompilieren.
```

Code used was the bare minimum, board Arduino Uno:

```
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

When deleting the hardware/avr in /home/mattias/.arduino15, everything works again. Is the problem due to my local setup or a bug in the updated toolchain?

Update, this weird hack seems to have fixed it (`$HOME/.arduino15/packages/arduino/hardware/avr/1.6.4/platform.txt`), I might check later for a proper fix.

```
# AVR compile variables
# ---------------------

compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra
```

matthijskooijman, 2015-04-24:
What is the hack exactly? You added the `compiler.warning_flags=-w` line I think?

@ffissore, so apparently the core 1.6.4 is not actually usable _at all_ without also upgrading the IDE. The fix suggested above would work as a backward compatibility thing, though if we add it, it should be clearly marked as such. Alternatively, releasing the 1.6.4 IDE and then marking the 1.6.4 core as needing the 1.6.4 IDE might be better (do we even have a way to mark this dependency now?)

mschlenker, 2015-04-24:
@matthijskooijman Yes, I added `compiler.warning_flags=-w` without further checking where this variable should get set. I assumed this might or might not get overwritten by the IDE so not causing too much harm.

matthijskooijman, 2015-04-24:
@mschlenker, the code in current git (which would be IDE 1.6.4) sets the `compiler.warning_flags` automatically, depending on preferences. Since this code isn't in 1.6.3 yet, your compile is breaking.

mschlenker, 2015-04-24:
@matthijskooijman This sounds reasonanble, but why is the toolchain released independently than?

matthijskooijman, 2015-04-24:
@mschlenker, for quicker and easier updates, I presume, but @ffissore knows more :-)

ffissore, 2015-04-24:
My mistake

mschlenker, 2015-04-24:
@ffissore Shit happens. It might make sense to pull back the 1.6.4 toolchain update, release it with a fixed platform.txt as 1.6.4.1 and then bring the complete update of IDE and toolchain as 1.6.5. This way no one has to manually delete the 1.6.4 toolchain.

ffissore, 2015-04-24:
I chose 1.6.5 since to adhere to semver. AVR core released. Please upgrade

closed 2015-04-24 by ffissore