arduino/Arduino #3031: Arduino.exe start, write initializing packages and closed.

VladimirKrd, 2015-04-24:
Do not work, Win8 x64. Java SE DK 8 Update 45

mschlenker, 2015-04-24:
Could you please check if #3015 applies to your problem?

VladimirKrd, 2015-04-24:
No, deleting a file does not work

VladimirKrd, 2015-04-24:
The program does not show any error messages, it just closes.

mschlenker, 2015-04-24:
Would you start it from the command line to check whether Java spills out some errors?

VladimirKrd, 2015-04-24:
Of course I can, you need to write in command line?

ffissore, 2015-04-24:
Use command prompt and `cd` into the Arduino IDE folder, then run `arduino_debug.exe`. It will print *stuff*: please paste it here

VladimirKrd, 2015-04-24:
[image]

mschlenker, 2015-04-25:
The missing `bootloaders/platform.txt` looks a bit like a broken install. Would you mind uninstalling and re-installing again?

VladimirKrd, 2015-04-25:
Ok. I downloaded the .exe, set does not work. Download *.zip, copied all, does not work. Now reinstall  *.exe, and all run! Fiction. Thank you so much for your help, God forbid you well.

mschlenker, 2015-04-25:
You might have a corrupted file system or bad hard drive. So you'd better run CHKDSK on your hard drives now. Have fun. And have a nice weekend!

closed 2015-04-25 by VladimirKrd

usef-96, 2015-05-04:
[image]
[image]
Hi,
I have the same problem but look in my case please can you help me .

ffissore, 2015-05-04:
Please delete file `library_index.json` from your `%APPDATA%\Arduino15\` folder. Then reopen the IDE

usef-96, 2015-05-04:
Thank you very very much mister I appreciate your help, have a good day.

Steltek, 2015-06-06:
Had the same problem because I installed over an old version. Fix was to delete the 'hardware' directory and re-extract it from the original ZIP archive.