arduino/Arduino #3032: ATtiny boards on 1.6.5

Lance-at85, 2015-04-25:
I have just upgraded from 1.5.2 to the 1.6.3 (updated boards to 1.6.5). 
I am using the Mini Pro 3.3v 8mHz, ATtiny 45 and ATtiny 85 and have been using version 1.0.5 for these boards but these do not appear in the Boards listings in version 1.6.3. 
Can I add the cores for these boards to 1.6.3 by just adding them to the C:\Program Files (x86)\Arduino\hardware\arduino folder.  -  Windows 7 x64bit
Or do I just keep using 1.0.5 for these particular boards/MCU's?

Regards.

Lance.

damellis, 2015-04-26:
In 1.6.3 you add boards to the same location as you did in 1.0.5, but the format of those boards folders and files has changed somewhat. If you're using my ATtiny folder (described here: http://highlowtech.org/?p=1695), you'll need the ide-1.6.x branch: https://github.com/damellis/attiny/tree/ide-1.6.x

For more details about the changes to the boards format, see: https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6

Lance-at85, 2015-04-26:
Many thanks for the information.
 I have downloaded the info you suggested and shall set up my ATtiny core files as recommended.

Lance-at85, 2015-04-27:
Damellis- 
Have included the ATtiny cores into the indicated folder and all are appearing in ARDUINO 1.6.3.
Now to work on the AT13 cores.
Thanks for your advice.

Lance.

closed 2015-05-01 by ffissore

wasimxe, 2016-05-09:
https://github.com/SpenceKonde/ATTinyCore work great for me, Thanks