arduino/Arduino #3041: SoftwareSerial sets pinMode in its constructor which is too soon

nickgammon, 2015-04-27:
In version 1.6.0 (at least) of the IDE the constructor for SoftwareSerial does this:

```
  setTX(transmitPin);
  setRX(receivePin);
```
setTX does this:

```
  pinMode(tx, OUTPUT);
  digitalWrite(tx, _inverse_logic ? LOW : HIGH);
```

This is not an appropriate time to be doing pinMode or digitalWrite because init() has not been called yet. Who knows what init will do with pins? It might set them all to inputs, for example.

mastrolinux, 2015-04-27:
related to: http://forum.arduino.cc/index.php?topic=318410.0

sbrynen, 2015-06-11:
I had a quick look, and I can't see why you couldn't just move the pinMode commands to the begin() method,  You need to call begin() to set the baud rate anyway before doing IO.

NeilMartin, 2018-11-02:
This problem can be dodged if you create your SoftwareSerial whenever you want the pins to be set. So you control when the constructor is called instead leaving it to global construction. For example, put this in your setup() function...

`mySoftwareSerial = new SoftwareSerial(pin_rx, pin_tx);`
`mySoftwareSerial->begin(baudRate);`

nickgammon, 2018-11-16:
Yes, that is a work-around, however it looks like the official version was fixed in commit 8c8b78ce4975103ed82067fe98f2101fcce48d68.

closed 2018-11-16 by nickgammon