arduino/Arduino #3054: Feature Request: command-line usage purely on CLI

jipp, 2015-04-29:
Hi

when using the command-line I would expect a pure command-line. - So no Window handling at all.

I tried it on OSX and Windows using 1.6.1 and nightly build.

After starting the command-line I see the Arduino logo popping up.
Also in case of an error (unknown option) the message this is given in a window not on the CLI.

To make the Arduino IDE more useful it would be better to handle everything purely on CLI.
This would allow the integration of the CLI into other projects, scripts, batch-jobs.

Maybe it would also be helpful to have some cli commands integrated like:
--version
--help (showing a list of available commands) - specially the documentation update is always behind and the help would might be solving this to some level

thanks for the good work you do.

Jipp

matthijskooijman, 2015-04-29:
The part about the splash screen and error popups is already reported at #1981, so let's focus this issue about adding --version and --help (which seem like a good idea to me). Btw, you say the documentation is always behind, but I actually think the commandline manpage is the single piece of documentation that is properly up-to-date always (since it is contained inside the git repository, so changes to the manpage can be merged alongside with the corresponding code changes).

jipp, 2015-04-29:
Hi

agreed.
The comment about documentation is just a general one. Its always hard to keep documentation and development aligned on the same level.

thx
Jipp

matthijskooijman, 2015-04-29:
Yup, totally :-)

mschlenker, 2015-04-30:
It gets worse, trying on a headless machine via SSH:

```
arduino16 --verify /home/mattias/Projekte/git/ManekiNeko2015/ManekiNeko/ManekiNeko.ino
```
results in:

```
Exception in thread "main" java.lang.ExceptionInInitializerError
Caused by: java.awt.HeadlessException: 
No X11 DISPLAY variable was set, but this program performed an operation which requires it.
	at java.awt.SplashScreen.getSplashScreen(SplashScreen.java:117)
	at processing.app.Base.<clinit>(Base.java:73)
```

Having experienced similar issues with running OpenOffice in batch mode for converting documents, I remembered to install the Xvfb X server and wrapping the whole arduino command in a shell script. If it's not possible to get rid of AWT in headless mode, at least the documentation should contain a hint. 

Should I add a section "ENVIRONMENT"  with a hint to DISPLAY and how to work ardound the issue to the man page and then pull request? https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

ffissore, 2015-04-30:
@mschlenker yes indeed that's the way we execute the tests on our linux box. We are developing a pure cli tool and I hope we'll release it in the coming months

mschlenker, 2015-04-30:
OK, I'll update the man page accordingly.

mschlenker, 2015-04-30:
See https://github.com/arduino/Arduino/pull/3062 - BTW thank you very much for #2904.

ffissore, 2015-05-29:
As for #1981, closing as wontfix. Fixing would require dismantling the IDE. We make small steps everyday, but can't really set a schedule

closed 2015-05-29 by ffissore