arduino/Arduino #3056: Arduino IDE environment colours

YorkieMan, 2015-04-29:
Very difficult to read the bottom notification display/error reporting panel (red on black background). Many of us have some colour sight impaiment !! Is it possible to change the colour scheme now, if so how? If not can you add some help for us ?

mschlenker, 2015-04-30:
#2901 and #2978 - this should be fixed in upstream. In the meantime you can follow @PaulStoffregen s suggestions for setting the colors (see the comments in the linked issues).

per1234, 2017-07-03:
>Very difficult to read the bottom notification display/error reporting panel (red on black background).

Done via https://github.com/arduino/Arduino/commit/8020c0d7339ede208f4a60934e65efb9a52b63df
> Is it possible to change the colour scheme now, if so how?

This was explained to some extent at https://github.com/arduino/Arduino/issues/2901#issuecomment-89553188.
>If not can you add some help for us ?

I'm interpreting this as a request for documentation of the theme settings, which I agree would be a good idea. It looks like we're tracking that request at https://github.com/arduino/Arduino/issues/6438

I think this issue is resolved so I'm closing it.

closed 2017-07-03 by per1234