arduino/Arduino #3058: Can I add libraries in which their repo name starts with a number in the new library manager?

iMustafa, 2015-04-29:
I know libraries directories can't start with a number,  what will happen if the name of the repo I'm adding starts with a number like:
https://github.com/Integreight/1Sheeld-Arduino-Library

Does the manager clone the repo with its current name? or there is a way we can rename the created directory?

ffissore, 2015-04-30:
What matters is the name of the library as written in `library.properties`

closed 2015-04-30 by ffissore

iMustafa, 2015-04-30:
As a test I tried to git clone the library in the Documents/Arduino/libraries directory and restarted Arduino.
I then restarted the IDE and I was faced by this error 4 times before the IDE starts.
[image]

The library name in library.properties file is "1Sheeld Arduino Library" the directory was "1Sheeld-Arduino-Library"

I then tried to rename the cloned directory to "OneSheeld" and left the name as it is in library.properties file the IDE started normally without any error.

So the directory name does matter I guess.

ffissore, 2015-04-30:
I'm sorry: I thought your question was related to the Library Manager.

If your library was listed in library manager, my first answer applies, because the lib manager creates the folder starting using the name specified in library.properties.

Since your cloning the repo into libraries folder, you're making the folder and it's up to you to use a valid name

iMustafa, 2015-04-30:
My question is indeed related to the Library Manager, but the library hasn't been listed yet. I am asking that question before the listing request.

Again, the name of the library in library.properties file is "1Sheeld Arduino Library" (It starts with a number) what is the implications of that after being listed in the Library Manager? What will be the name of the library directory? will it be the name from library.properties or the repo name?

ffissore, 2015-04-30:
library.properties

iMustafa, 2015-04-30:
@ffissore Thanks, just to make sure I got it, the name inside library.properties should not begin with a number, correct?

ffissore, 2015-04-30:
correct