arduino/Arduino #3059: Assembler No Longer Support

Coding-Badly, 2015-04-29:
- Save a bare minimum sketch.
- Add a dot-S file to the folder; the file can be empty.
- Close then reopen the IDE.
- Open the recently saved bare minimum sketch.
- Verify.

The IDE makes no attempt to include the assembler file in the build.  The platform.txt file includes entries for dot-S files implying they are supported.

Additional information: With the IDE still open...
- Click the down-arrow on the far right side of the window.
- Click New Tab.
- Enter a filename with a dot-S suffix.
- The IDE complains that: ".s" is not a valid extension.

PaulStoffregen, 2015-04-30:
Arduino has never supported .s assembler files in sketches.  They are supported in libraries and the core library, but not sketches.

I have a patch that adds this support for sketches.  If the Arduino devs say they'd like to add this, I'd be happy to prepare a pull request.

This has been discussed before.  I believe the decision, by David Mellis, was not to support .s files in sketch.  As a general rule, I don't put time into pull requests for features that have been previously rejected.  I just maintain the patches and update them for each Arduino release.  Of course, I'm happy to contribute if the feature is wanted.

Coding-Badly, 2015-04-30:
Thank you for the post!

Huh.  I know René Bohne used some assembly in LumiNet.  For some reason I got it my head the code was delivered as a sketch but that cannot possibly be true.  LumiNet used the t84 processor.

In any case, there is a person on the forum who really wants assembly support.  I will suggest packaging the assembly as a library.

q2dg, 2015-04-30:
Yes. If it's possible and it's done, why not?

PaulStoffregen, 2015-04-30:
Maybe @damellis could explain his thinking about why Arduino doesn't support .s files in sketches?

I would agree with "why not".  In fact, I put this patch into my collection of unofficial patches for Teensy, because a few different people requested it.  If you want to use .s files in a sketch, you can install Teensyduino.  Most of my patches check if a Teensy board is selected and don't alter Arduino's behavior, but for this one I didn't bother with the extra checking.  It should allow using .s file in sketches for any type of board.

PaulStoffregen, 2015-04-30:
Even without the .s file support, you CAN use inline assembly.  It's a different coding style that many people who want to just use assembly don't like, but really it's about the same if you merely create empty functions to serve as containers for assembly.

ffissore, 2015-05-26:
Fixed in #3227. Available since next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-05-26 by ffissore