arduino/Arduino #3060: NOT BE ABLE TO UPLOAD bug: Arduino 1.6.4. =>"upload with programmer" (AVR-ISP-MK2 from Olimex)

jack-zande, 2015-04-30:
Hello all,

Using new Arduino Software 1.6.4 (IDE) and an external programmer AVR-ISP-MK2 (Olimex)
on the ISP-connector on an Arduino UNO board, it can't upload sketches in the microcontroller.
Arduino Software can't see any connection with this progammer.

The new version 1.6.4 doesn't seem  to support the "Upload with programmer" with AVR-ISP-MK2
any more. 

Although when I use  Arduino version 1.0.5 it works fine.

Jack.

ffissore, 2015-04-30:
Please paste the error message

jack-zande, 2015-04-30:
Hi,

Error messages:

avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_command(): failed miserably to execute command 0x11
avrdude: stk500v2_disable(): failed to leave programming mode

ffissore, 2015-04-30:
Have you tried asking on the [arduino forum](http://forum.arduino.cc/) or the [olimex forum](https://www.olimex.com/forum/)?

jack-zande, 2015-04-30:
Hi ffissore,

Arduino forum, yes       > advise: reporting a bug  to ARDUINO support
olimex forum, no           > there is no newer firmware update for the AVR-ISP-MK2, but it works 
                                       fine with ARDUINO 1.0.5   (others, older ones, I didn't use)
                                       (driver hex-file: used from Olimex <libUSB-AVRISP-MKII.hex>)
Jack

jack-zande, 2015-04-30:
driver hex-file:

libUSB-AVRISP-MKII.hex

ffissore, 2015-04-30:
ide 1.0 ships avrdude 5.11 while ide 1.6 ships avrdude 6.0.1. you may want to ask support to olimex and get advice about avrdude version. maybe their programmer just doesn't work with anything newer than 5.11, so you're stuck with ide 1.0.5, or maybe it's something else I can't help you with

jack-zande, 2015-04-30:
Yes, maybe Olimex is behind new developments. Thanks for the info.
I should report this to them and hope they are interested.

You are right, I am stuck with V1.0.5. Can't work with OLIMEX AVR-ISP-mk2 together.

Although I can use STUDIO 4 to program the UNO on the ISP. No problems there.

Jack

ffissore, 2015-04-30:
Thank you for reporting. Please keep us posted. If a patched or newer avrdude fixes the problem, we may work on that

ffissore, 2015-08-06:
If olimex has issues with recent avrdude, they should provide their own core with their own avrdude version. Closing as invalid as this is not a bug we can fix on our own

closed 2015-08-06 by ffissore