arduino/Arduino #3063: LCD library display code does not position text properly when compiled with latest IDE, but works with 1.0.6

PietBuilder, 2015-04-30:
// engine condition monitoring routines
  // cylinder head temperatures, No 1 to 4

// Define the number of samples to keep track of.  The higher the number,
// the more the readings will be smoothed, but the slower the output will
// respond to the input.  Using a constant rather than a normal variable lets
// use this value to determine the size of the readings array.
const int numReadings = 20;
const int number_of_Analogues_in_use = 4;

// start of LCD pin definitions
#include <LiquidCrystal.h> // load the LCD library
  int LCD_RS_pin     = 7;    // was 12
  int LCD_Enable_pin = 8;    // was 11
  int LCD_D4_pin     = 9;    // was 5
  int LCD_D5_pin     = 10;   // was 4
  int LCD_D6_pin     = 11;   // was 3
  int LCD_D7_pin     = 12;   // was 2
LiquidCrystal lcd(LCD_RS_pin, LCD_Enable_pin, LCD_D4_pin, LCD_D5_pin, LCD_D6_pin, LCD_D7_pin);
  // ---------------------------------------------------------
  // start of alarm variable and startup variable definitions
  // ---------------------------------------------------------
int power_up_has_occurred            =  true;
int No1_cht_temperature_is_high      =  false;
int No2_cht_temperature_is_high      =  false;
int No3_cht_temperature_is_high      =  false;
int No4_cht_temperature_is_high      =  false;
int first_run_flag                   =  false;
int alarm_led_output                 =  13;    // engine monitoring module alarm LED is connected to this output
  // ---------------------------------------------------------
  // end of alarm variable and startup variable definitions
  // ---------------------------------------------------------
int start_of_blanking_column_number = 14;
int single_digit_column_number = 17;
int double_digit_column_number = 16;
int triple_digit_column_number = 15;

int readings_No1_cht[numReadings];
int readings_No2_cht[numReadings];
int readings_No3_cht[numReadings];
int readings_No4_cht[numReadings];

int No1_cht_index = 0;                  // the index of the current reading
int No2_cht_index = 0;                  // the index of the current reading
int No3_cht_index = 0;                  // the index of the current reading
int No4_cht_index = 0;                  // the index of the current reading

int total_No1_cht = 0;                  // the running total
int total_No2_cht = 0;                  // the running total
int total_No3_cht = 0;                  // the running total
int total_No4_cht = 0;                  // the running total

int average_No1_cht_temperature = 0;    // the average
int average_No2_cht_temperature = 0;    // the average
int average_No3_cht_temperature = 0;    // the average
int average_No4_cht_temperature = 0;    // the average

int No1_cht_temperature_sample = A0;
int No2_cht_temperature_sample = A1;
int No3_cht_temperature_sample = A2;
int No4_cht_temperature_sample = A3;

int No1_cht_temperature_to_display = 0;
int No2_cht_temperature_to_display = 0;
int No3_cht_temperature_to_display = 0;
int No4_cht_temperature_to_display = 0;

int cylinder_head_alarm_temperature = 270; // max permitted temp = 273c or 525f
  // end of variable definitions
  // ----------------------------------------------------
[... 357 lines cut ...]
  //------------------------------------------------
  // end of No4 CHT display code
  // -----------------------------------------------
  // --------------------------------------------------------------
  // start of alarm combining routine
  // -------------------------------------------------------------- 
if (No1_cht_temperature_is_high || No2_cht_temperature_is_high || No3_cht_temperature_is_high || No4_cht_temperature_is_high)
{
    digitalWrite(alarm_led_output, HIGH);
    delay(100);
    digitalWrite(alarm_led_output, LOW);
    delay(100);
}
else
{
    digitalWrite(alarm_led_output, LOW);
}
  // -------------------------------------------------------------
  // end of alarm combining routine
  // -------------------------------------------------------------  
  
  }
// curlywig on line above is the closing bracket for
// the end of main loop which scannes and writes out the data
//----------------------------------------------------------

cmaglie, 2015-04-30:
@PietBuilder 
what's the expected output? and what you actually get from this sketch?

PietBuilder, 2015-04-30:
Hello,

The sketch drives a 4 by 20 LCD display.

The title text for each row is written to the LCD once and the dynamic 
values are then contiously updated during each software cycle.

The inputs signals are captured from analog inputs 1,2,3 and 4 which 
have 250 ohm resistors to 0V on each input and each input is feed with 4 
to 20mA from a type K thermocouple converter.

Each input scaled to zero to 800 centigrade and the resulting value is 
tested for size (1,2 or 3 digits) and the cursor is suitably positioned 
prior to writing the value to the LCD.

When compiled with IDE 1.0.6 all is well (see attached photographs), 
however when compiled with  1.6.3 the text for the first two rows is 
correct, but the static text and the dynamic values for bottom two rows 
is written in very strage places, including in places in the first two rows.

Keith

cmaglie, 2015-05-07:
I can't see the pictures, you must use the web-interface of github to post them.

BTW you said that your display is 4x20 but in the setup you did `lcd.begin(16,4);`, maybe changing it to `lcd.begin(20,4);` solves your problem?

PietBuilder, 2015-05-09:
Thank you so much, that has solved my problem.

Keith

closed 2015-05-10 by cmaglie