arduino/Arduino #3067: Add SparkFun TLC5940 library

ToniCorinne, 2015-04-30:
Request to add SparkFun TLC5940 to master library list in library manager.

Repo is [here](https://github.com/sparkfun/SparkFun_TLC5940_Arduino_Library).

Most recent tag is [V_1.0.0](https://github.com/sparkfun/SparkFun_TLC5940_Arduino_Library/tree/V_1.0.0)

Thanks!

ffissore, 2015-04-30:
Added. Thanks!

closed 2015-04-30 by ffissore

PaulStoffregen, 2015-04-30:
This appears to be duplicate (and older) copy of the Tlc5940 library already added, with some non-library files added.

ToniCorinne, 2015-04-30:
Are we not allowed to add our own versions if we will be updating them and modifying them to work for our own projects?  We've got several examples for two different projects based on this library, so if that's the case, I need to let our product code folks know.

ffissore, 2015-04-30:
There also is an issue with the library.properties file: `'author' and 'maintainer' fields must be defined,'sentence' and 'url' fields must be defined`

ffissore, 2015-04-30:
I guess the problem is wit the blank before the = here: https://github.com/sparkfun/SparkFun_TLC5940_Arduino_Library/blob/master/library.properties#L3

ToniCorinne, 2015-04-30:
Ah, right. Will get that removed. Sorry about that.

PaulStoffregen, 2015-04-30:
As far as I know, there doesn't seem to be any published policy or even guidance on library duplication.

But this is a small issue that's likely to become larger over time.  For example, I saw another library from Adafruit which is merely an older copy of OneWire, with an example added for a specific sensor they're selling.

I can understand how companies like Sparkfun and Adafruit have a strong commercial interest in publishing their own copies of libraries.  But as someone who maintains several of these older libraries, and has for years, I'm not so excited about seeing stagnent forks of code I maintain, especially when the code itself appears to have no additions or improvements other than vendor-specific examples or supplementary files added.