arduino/Arduino #3069: nullpointexception error when I try to open preferences

kayosetsu, 2015-04-30:
Just installed be Arduino IDE but whenever I try to open the preferences settings I get the following error:

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.sun.java.swing.plaf.windows.XPStyle.getTypeEnumName(XPStyle.java:155)
	at com.sun.java.swing.plaf.windows.XPStyle.getBorder(XPStyle.java:277)
	at com.sun.java.swing.plaf.windows.WindowsLookAndFeel$XPBorderValue.getXPValue(WindowsLookAndFeel.java:2402)
	at com.sun.java.swing.plaf.windows.WindowsLookAndFeel$XPValue.createValue(WindowsLookAndFeel.java:2357)
	at javax.swing.UIDefaults.getFromHashtable(UIDefaults.java:233)
	at javax.swing.UIDefaults.get(UIDefaults.java:163)
	at javax.swing.MultiUIDefaults.get(MultiUIDefaults.java:64)
	at javax.swing.UIDefaults.getBorder(UIDefaults.java:481)
	at javax.swing.UIManager.getBorder(UIManager.java:759)
	at javax.swing.border.TitledBorder.getBorder(TitledBorder.java:416)
	at javax.swing.border.TitledBorder.getBorderInsets(TitledBorder.java:349)
	at javax.swing.border.AbstractBorder.getBorderInsets(AbstractBorder.java:75)
	at javax.swing.JComponent.getInsets(JComponent.java:1824)
	at javax.swing.GroupLayout.adjustSize(GroupLayout.java:1109)
	at javax.swing.GroupLayout.minimumLayoutSize(GroupLayout.java:897)
	at java.awt.Container.minimumSize(Container.java:1846)
	at java.awt.Container.getMinimumSize(Container.java:1830)
	at javax.swing.JComponent.getMinimumSize(JComponent.java:1744)
	at javax.swing.GroupLayout$ComponentSpring.calculateNonlinkedMinimumSize(GroupLayout.java:3008)
	at javax.swing.GroupLayout$ComponentSpring.calculateMinimumSize(GroupLayout.java:2972)
	at javax.swing.GroupLayout$Spring.getMinimumSize(GroupLayout.java:1336)
	at javax.swing.GroupLayout$Group.getSpringSize(GroupLayout.java:1636)
	at javax.swing.GroupLayout$Group.calculateSize(GroupLayout.java:1622)
	at javax.swing.GroupLayout$Group.calculateMinimumSize(GroupLayout.java:1598)
	at javax.swing.GroupLayout$Spring.getMinimumSize(GroupLayout.java:1336)
	at javax.swing.GroupLayout$Group.getSpringSize(GroupLayout.java:1636)
	at javax.swing.GroupLayout$Group.calculateSize(GroupLayout.java:1622)
	at javax.swing.GroupLayout$Group.calculateMinimumSize(GroupLayout.java:1598)
	at javax.swing.GroupLayout$ParallelGroup.calculateMinimumSize(GroupLayout.java:2569)
	at javax.swing.GroupLayout$Spring.getMinimumSize(GroupLayout.java:1336)
	at javax.swing.GroupLayout$Group.getSpringSize(GroupLayout.java:1636)
	at javax.swing.GroupLayout$Group.calculateSize(GroupLayout.java:1622)
	at javax.swing.GroupLayout$Group.calculateMinimumSize(GroupLayout.java:1598)
	at javax.swing.GroupLayout$Spring.getMinimumSize(GroupLayout.java:1336)
	at javax.swing.GroupLayout.minimumLayoutSize(GroupLayout.java:897)
	at java.awt.Container.minimumSize(Container.java:1846)
	at java.awt.Container.getMinimumSize(Container.java:1830)
	at javax.swing.JComponent.getMinimumSize(JComponent.java:1744)
	at processing.app.Preferences.<init>(Preferences.java:431)
	at processing.app.Base.handlePrefs(Base.java:1645)
	at processing.app.Editor$12.actionPerformed(Editor.java:610)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3322)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2739)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:751)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:702)
	at java.awt.EventQueue$3.run(EventQueue.java:696)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:721)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

I have reinstalled it multiple times but I still get the error. Can anyone please explain why does it hate me?

mschlenker, 2015-05-01:
https://netbeans.org/bugzilla/show_bug.cgi?id=183999 You are using XP? With a custom style vor the default style? Could you check whether the NPE also appears when you switch to the Classic (win2k) style?

mschlenker, 2015-05-04:
[image]

[image]

I dug out a VM running a relatively clean install of XP - with the latest patches so far. No problems. Probably an incompatible theme? BTW IE8 did not allow me to download Arduino, I had to install firefox first.

Looked like some impressions from a long distant past to me. XP is deprecated since more than one year and Arduino 1.6.3 even works with a quite vanilla XP, so I guess this issue can be closed.

closed 2015-05-07 by ffissore