arduino/Arduino #3072: Where find the source of Functions of StringObject

utix, 2015-05-01:
Hi,

I would like to help, and improve the documentation.
For exemple this one:

http://www.arduino.cc/en/Reference/StringToCharArray

I can't find the doc, I have search into https://github.com/arduino/reference-en
Where can I get it and submit some patches ?

This function add the final '\0' and triggers question like this http://stackoverflow.com/questions/29990995/arduino-switch-with-chars

Where can we find the implementation of this function ?

BR

DraakUSA, 2015-05-01:
If you read near the bottom of the page...

*Corrections, suggestions, and new documentation should be posted to the [Forum](http://arduino.cc/forum/index.php/board,23.0.html).*

utix, 2015-05-02:
Great, I propose to help, dude !!

 @DraakUSA if you read the forum, you will found an important notice.
And......... What you will read:

- Bug reports and simple enhancement requests should go on the GitHub issue list:
http://github.com/arduino/Arduino/issues

Good ?
So this is an simple enhancement, just update the doc. And I propose do to it. I have found the function but not the documentation

I am just asking for the repository.

q2dg, 2015-05-02:
Well, it seems documentation is here: https://github.com/arduino/reference-en/tree/f109a699dbbfdbb46a63a6203a3610feac6bc4a3/Language

But if you go to "Functions" category, there's no "String" documentation (there is a "Serial" or a "Stream" section, for instance, but not "String"). The only related thing is https://github.com/arduino/reference-en/blob/f109a699dbbfdbb46a63a6203a3610feac6bc4a3/Language/Variables/Data%20Types/stringObject.adoc , which points to web...

A little messy

utix, 2015-05-02:
I have try to find it but I can't.
The function is defined here:
* [/hardware/arduino/avr/cores/arduino/WString.h](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/WString.h)

* [/hardware/arduino/sam/cores/arduino/WString.h](https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/WString.h)

agdl, 2015-05-08:
@utix thank you for the support. If you want to update and improve the documentation you are very welcome if you wanna make a PR (or open an issue) in the repo "reference" you linked. Unfortunately the online doc doesn't point to that repo yet, but if some mod is made there i will update the web pages accordingly!

agdl, 2015-06-05:
I'm closing this. Reopen it if further doubts are present

closed 2015-06-05 by agdl