arduino/Arduino #3075: Arduino Playgrond readonly?  Unable to edit article I wrote (a few years ago)

wayoda, 2015-05-03:
A few years ago I wrote an article on the arduino playground about one of my libraries:
http://playground.arduino.cc/Main/LedControl

Since then the [sourcecode](https://github.com/wayoda/LedControl) and the [documentation](http://wayoda.github.io/LedControl/) moved to github. 

Today I tried to remove the old doc from the playground and replace it with link to the new location, but the playground seems to be readonly for normal users? When clicking the edit link there is a prompt for a password even though I'm signed in as a forum user (which was ok last time I tried to edit something on the playground)

ffissore, 2015-05-04:
/cc @mastrolinux

mastrolinux, 2015-05-07:
Hello,

Please send us an email at webmaster@arduino.cc and we will give you the password.

wayoda, 2015-05-08:
I'll do that, thanks!
Eberhard

closed 2015-05-08 by wayoda