arduino/Arduino #3076: playground.arduino.cc/learning/linux needs update

dplasa, 2015-05-03:
It states

>If you want to use your system's compiler, delete the folder ./hardware/tools/avr in your arduino IDE 
>installation 

which is not good anymore as arduino will crash with a null pointer exception after this. This is a bug and should be fixed. However also the documentation should be updated to

*If you want to use the system avr-gcc, the best to do so is to modify platform.txt, or to add a platform.local.txt alongside it. It should contain:

    compiler.path=
    tools.avrdude.cmd.path=avrdude
    tools.avrdude.config.path=/etc/avrdude.conf

ffissore, 2015-05-04:
We won't change documentation: we'll fix the bug instead. See #3074.

closed 2015-05-04 by ffissore

dplasa, 2015-05-05:
That's even better :)

However mentioning the files
    platform.txt
    platform_local.txt
in this part of documentation would also be nice.