arduino/Arduino #3080: Add sketch subdir in the Include Path

Testato, 2015-05-05:
Actually is possible include in the sketch an header file only if it is in:

pippo\pippo.ino
pippo\x.h
c:\x.h

Please add the possibilities of use subdir in the sketch:

pippo\somedir\x.h

ffissore, 2015-05-05:
Can you also code and [put on gist](http://gist.github.com/) a short example sketch that we can use to test the feature?

MauroMombelli, 2015-05-05:
see #1152, the sdlib is a perfet example that actually works because a special case in the code

matthijskooijman, 2015-05-05:
Seems you might be mixing up things. For "new-style" libraries, source files in subdirectories of the src/ directory are compiled, but AFAIK these subdirectories are _not_ added to the include path. Doing so would actually be counter-productive - it would merge all those directories in a single namespace, greatly increasing the chance of naming conflicts.

If you want to include pippo\somedir\x.h from pippo.ino, you can just use `#include "somedir/x.h"`, which is nice and explicit, without relying on any magic include path. Are you not aware of this option, or is there something wrong with this option? Perhaps you mean something entirely different - you should really provide a more verbose explanation and/or  examples...

Testato, 2015-05-05:
#include "somedir/x.h"

this is exactly the think that do not work, i do not skeak about a library problem, but simply including header like your example. This is very usefull on big project.

I uploaded a simple sketch+subdir here:
https://drive.google.com/file/d/0Bw27MYXTb2FXQ0kxZXdMQkpmajg/view?usp=sharing

matthijskooijman, 2015-05-05:
Could it be a \ vs / issue? I think includes normally use /, even on Windows. Try using a / in your include too and tell use if that helps.

Testato, 2015-05-05:
same problem whit \ or /

Testato, 2015-05-05:
no, the problem is the same.

On Arduino IDE, at least on Windows but i think on linux also, there is no differenze:

you can do this:
#include "C:/test.h"

or this
#include "C:\test.h"

Try yourself whit my uploaded sketch

Testato, 2015-05-05:
this is the example in the uploaded file:

```C++
#include "test1.h"	    // This work
#include "C:\test2.h"       // This work
#include "SomeDir\test3.h"  // This is the ISSUE request

void setup() {
}

void loop() {
}
```

matthijskooijman, 2015-05-05:
Ah, I understand what happens now - .ino files are preprocessed and the result is written to the temporary build directory. I think any .c / .cpp and .h files are copied to the build dir too, but not subdirectories, so the include does not work (just tested this, breaks for me too).

To fix this we could put the original sketch directory (not the subdirectory) in the include path. However, this blurs the distinction between "" and <> includes, so I'm a bit hesitant to propose that. Copying subdirectories into the build dir might make sense, though then I guess also .cpp and .c files in subdirectories should be compiled for consitency (might be a good idea anyway).

@ffissore, any thoughts?=

Testato, 2015-05-05:
I vote for copying subdirectories into the build dir together whit any .h .c .cpp inside the subdir

ffissore, 2015-05-05:
I think this is related to #2997. The same reasons for fixing #2997 apply here. A safe way for fixing both could be to 1) create three subfolders in the temp dir: "core", "libraries", "sketch" 2) preserve the whole directory structure for each
This would allow a (prefixed) 1 to 1 mapping between sources and object files
Short example
```
/tmp/build123/core/cores/arduino/WString.cpp.d
/tmp/build123/core/cores/arduino/WString.cpp.o
/tmp/build123/libraries/Ethernet/Ethernet.cpp.d
/tmp/build123/libraries/Ethernet/Ethernet.cpp.c
/tmp/build123/sketch/main.cpp.o
/tmp/build123/sketch/main.cpp.d
/tmp/build123/sketch/sketch_may05a.cpp
/tmp/build123/sketch/SomeDir/test3.h
```

I wonder if such a structure could make life even harder WRT #1337

Testato, 2015-05-05:
this new structure can only be a good thing :-)
but if it need too many work please consider to start to add only SomeDir/test3.h feature, if you search on google on this topic you found infinite post whit infinite failed solution  because nobody beleave that it is not possible use an heder file under a subdir :-)

MauroMombelli, 2015-05-05:
@ffissore this is a good starting point, there i see 2 main solution; one is to slowly break compatibility with that system and start inducing user to use relative path (the 3 "entry point" are core/cores/arduino/, libraries, and tmp folder)

or keeping the compatibility but using only one gcc command, you can stop coping everything and just copy all the file but the .ino, generate the sketch_may05a.cpp, the ONE gcc command like

cd /tmp/build123/
gcc sketch_may05a.cpp $include

where $include is a list of absolute path to ALL folder into the "libray" plus "core/cores/arduino/" (obviusly depend on the core used).

OR

you can even skip the copy of the subfolder of the sketch, and add it as the include; the nice thing is that the include orader is the PRIOORITY order, so if multiple .h are present (like in the case you are using the IRlibrary) we don't compilation error; i think the sketck folder should win again all, themn libraries, and then core. 
This is good if for some reason someone or some library want to override some code/library code.

Testato, 2015-06-29:
Now that #2997 is cosed, is it possible close this isdue also ?

ffissore, 2015-06-29:
here we go #3435

closed 2015-06-30 by ffissore

Testato, 2015-06-30:
Very good, i will test it.
Thanks

PR #3435 (2015-06-30): Recusively copying additional files
Compiler: additional files with allowed extentions are recursively copied to build folder. Fixes #3080
commit ecb4eaf8 (2015-06-30): Compiler: additional files with allowed extentions are recursively copied to build folder. Fixes #3080