arduino/Arduino #3081: Feature Request: Support for jayatanaag (global menus in Ubuntu)

neopostmodern, 2015-05-05:
(This was previously filed as bug https://github.com/arduino/Arduino/issues/3024)

For Ubuntu users [Java Swing Jayatana](https://code.google.com/p/java-swing-ayatana/) _"is a project that brings Ubuntu global menu (AppMenu) and HUD support for Java Swing applications."_ (says [webupd8.org](http://www.webupd8.org/2014/02/get-unity-global-menu-hud-support-for.html)).

On both Ubuntu 14.10 & 15.04 (possibly more versions, especially 14.04 - but not confirmed) Arduino IDE will crash under certain circumstances (namely `Save`/`Save As`) with jayatanaag enabled. Disabling it fixes the issue.

Since all other, major software products [that I personally use] run fine with jayatanaag there might be something that could be improved in Arduino IDE to fix this behavior.

**Technical**
*On Ubuntu 14.10/15.04 x64, Java 1.7.0_80*
```plain
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f74cbe94c84, pid=27872, tid=140137198511872
#
# JRE version: Java(TM) SE Runtime Environment (7.0_80-b15) (build 1.7.0_80-b15)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.80-b11 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libpthread.so.0+0x9c84]  pthread_mutex_lock+0x4
#
# Core dump written. Default location: /usr/share/arduino/core or core.27872
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread is native thread

siginfo:si_signo=SIGSEGV: si_errno=0, si_code=1 (SEGV_MAPERR), si_addr=0x0000000000000010

Registers:
RAX=0x00007f7495a7d9b0, RBX=0x00007f74c44b1640, RCX=0x000000000000014c, RDX=0x00007f743bef9dd0
RSP=0x00007f743bef9d60, RBP=0x00007f743bef9de0, RSI=0x00007f743bef9dc0, RDI=0x0000000000000000
R8 =0x00007f743bef9de0, R9 =0x00007f7495b00487, R10=0x0000000000000065, R11=0x0000000000000004
R12=0x00007f743bef9dbc, R13=0x0000000000000000, R14=0x00007f74c4616258, R15=0x00007f743befb340
RIP=0x00007f74cbe94c84, EFLAGS=0x0000000000010206, CSGSFS=0x0000000000000033, ERR=0x0000000000000004
  TRAPNO=0x000000000000000e

Top of Stack: (sp=0x00007f743bef9d60)
0x00007f743bef9d60:   00007f74c4615cc0 00007f7495a9756a
0x00007f743bef9d70:   00007f743bef9dc0 00007f743bef9dd0
0x00007f743bef9d80:   00007f746cc85be1 00007f746cc87414
0x00007f743bef9d90:   00007f74c4434a70 00007f746cc85be1
0x00007f743bef9da0:   00007f746cc87414 00007f7495a74b36
0x00007f743bef9db0:   00007f744c09ea40 00007f742c0848e0
0x00007f743bef9dc0:   0000014d0000014c 0000000000000000
0x00007f743bef9dd0:   0000015500000154 00007f7400000000
0x00007f743bef9de0:   00007f74c4434a70 00007f74c4614af8
0x00007f743bef9df0:   00007f74c4615cc0 00007f74c4615cf4
0x00007f743bef9e00:   00007f74c4434a70 00007f746cc3fe7f
0x00007f743bef9e10:   0000000000000000 00007f746c5312b0
0x00007f743bef9e20:   00007f74c4614ad0 f29d4b0380acee00
0x00007f743bef9e30:   0000000000000000 00007f742c07ece8
0x00007f743bef9e40:   00007f743bef9ea0 00007f742c07ece8
0x00007f743bef9e50:   0000000000000000 00007f74c4616258
0x00007f743bef9e60:   00007f743befb340 00007f746cc42420
0x00007f743bef9e70:   00007f742c07ebc0 00007f746cc13027
0x00007f743bef9e80:   00007f74c4616258 00007f742c0492e0
0x00007f743bef9e90:   00007f743bef9ec0 00007f746cbd107c
0x00007f743bef9ea0:   00007f744c0102a0 00007f746c777593
0x00007f743bef9eb0:   40cd555555555555 0000000000000065
0x00007f743bef9ec0:   0000000000000028 00007f743befa010
0x00007f743bef9ed0:   00007f743bef9fa8 00007f743befa090
0x00007f743bef9ee0:   00007f7462d27fb8 0000000000000028
0x00007f743bef9ef0:   00007f7462d28744 f29d4b0380acee00
0x00007f743bef9f00:   00007f74c4616090 00007f742c0492e0
0x00007f743bef9f10:   0000000000000050 00007f746cbd4d83
[... 1095 lines cut ...]
cache size	: 6144 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm ida arat epb xsaveopt pln pts dtherm tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms
bogomips	: 5387.53
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

Memory: 4k page, physical 15996316k(8824912k free), swap 0k(0k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (24.80-b11) for linux-amd64 JRE (1.7.0_80-b15), built on Apr 10 2015 19:53:14 by "java_re" with gcc 4.3.0 20080428 (Red Hat 4.3.0-8)

time: Tue May  5 10:39:04 2015
elapsed time: 5 seconds
```

SurferTim, 2015-05-06:
This can be corrected by a simple startup script edit.
http://forum.arduino.cc/index.php?topic=318520.0

matthijskooijman, 2015-05-06:
So IIUC, you say adding `export JAVA_TOOL_OPTIONS=""` to the startup script fixes this? Any idea why this works and what this variable does? I think the devs will need a bit more info on what this does (and possible side effects) before this can be changed as the default.

ffissore, 2015-05-06:
[JAVA_TOOL_OPTIONS](https://docs.oracle.com/javase/7/docs/webnotes/tsg/TSG-VM/html/envvars.html) is a jvm env var ubuntu uses to inject a javaagent, a thing that the jvm puts between application code and its actual execution. jayatana uses this trick to change how menu are rendered. Unfortunately, jayatana is buggy and unmaintained.
The suggested fix is therefore good, especially because it only affects Arduino

matthijskooijman, 2015-05-06:
Hm, if that software is indeed buggy and unmaintained, I would think this is not a proper fix - people should stop running it instead (or fix it). By adding this work around, _all_ injected java agents will be disabled, even others that are not broken. This might lead to unexpected behaviour.

Disabling this variable is probably the most practical thing to do, though. Perhaps we can only clear the variable when it contains a reference to jayatana and leave it intact otherwise?

SurferTim, 2015-05-06:
Yes, it fixes it. I'm using it with IDE versions 1.0.5 and 1.6.3.

You can run the IDE as root and it does not generate the
"Picked up JAVA_TOOL_OPTIONS: -javaagent:/usr/share/java/jayatanaag.jar "
line when run from a terminal, and it runs fine.

If you use this from a terminal as a user
export JAVA_TOOL-OPTIONS=""
./arduino

it generates this line
"Picked up JAVA_TOOL_OPTIONS:"
and the IDE runs normally.

Tim

SurferTim, 2015-05-06:
BTW, using the edited arduino script does not change the environment variable permanently. When you exit the IDE, the original default value shows up again. From a command terminal after exiting the IDE that has the edit:

echo $JAVA_TOOL_OPTIONS
 -javaagent:/usr/share/java/jayatanaag.jar

neopostmodern, 2015-05-07:
I filed this as a feature request, not a bug report (there was https://github.com/arduino/Arduino/issues/3024), so the workaround isn't quite what I was looking for.
I personally like *jayatana*'s improvements on usability and was hoping I could get them in arduino too.

I am pretty confident *jayatana* isn't being discontinued since it just got promoted into the regular (`release`) Ubuntu repositories: http://www.webupd8.org/2015/01/ubuntu-1504-might-get-global-menu-and.html - This would quite to the contrary increase the importance of the feature for regular ubuntu users.

I'll check arduino's compatibility with the `release` repository's *jayatana* but I guess behavior will be the same.

SurferTim, 2015-05-07:
My bad. I'll post on the bug report.

What features does jayatana offer above what Java already has?

neopostmodern, 2015-05-07:
As stated above in the original post it says:

> For Ubuntu users [Java Swing Jayatana](https://code.google.com/p/java-swing-ayatana/) _"is a project that brings Ubuntu global menu (AppMenu) and HUD support for Java Swing applications."_ (says [webupd8.org](http://www.webupd8.org/2014/02/get-unity-global-menu-hud-support-for.html)).

closed 2015-05-12 by ffissore

neopostmodern, 2015-05-12:
@ffissore this fixes the bug #3024 - but it doesn't add the feature? Or is the feature request somehow invalid?

ffissore, 2015-05-12:
ATM supporting jayatana would mean digging into its code and find the reason for its crash. It's rather something that should be reported upstream https://code.google.com/p/java-swing-ayatana/issues/list
Would you like to open jayatana an issue?

lmihalkovic, 2015-12-22:
[LMCHK]

commit 7c5c1741 (2015-05-12): GNU/Linux: removing jayatana from JAVA_TOOL_OPTIONS, if present. Fixes #3081