arduino/Arduino #3086: "file -> Examples " menu bigger than the screen

cyberlink1, 2015-05-05:
I have a lot of add-on libraries and noticed that the File --> Examples menu is bigger than the screen and there is no way to get to the ones that are above or below the screen height.

Arduino 1.6.3/ on Linux

PaulStoffregen, 2015-05-06:
What about clicking the little arrows at the top & bottom of the list?  Or using the scroll wheel on your mouse?  Both of those work for me with 1.6.3 on Ubuntu 14.04.  Do neither work on your Linux system?

cyberlink1, 2015-05-06:
No, I dont have the little arrows and the scroll wheel does nothing.

Im on Fedora Core 

[image]

PaulStoffregen, 2015-05-06:
On Ubuntu, the top and bottom of the menu have little arrows, and the mouse scroll wheel scrolls the whole thing uo/down.

Maybe there are TWO bugs on Fedora, causing incorrect screen height reporting to Java, AND not allowing the mouse scroll wheel to work?

I just tested again here, and the keyboard arrow keys also work, causing the entire menu to scroll when pressing the key to the position past the last item.  Maybe *that* can work?

In any case, I'm pretty sure you're experiencing bugs due to Fedora, not Arduino.  It works great on Ubuntu, and Windows and Mac.

cyberlink1, 2015-05-06:
I could see it being a java version issue, or a bug in java but not really a bug in the linux distribution.

I did track the issue down and identify the root cause.

I have two monitors, one was at 1360x768 (Primary) and one was at 1600x900 (Secondary) When I make both monitors the same resolution (1360x768) the arrows show up and everything works as expected. When the monitors are at different resolutions it appears to select the larger monitor as the size settings for the scroll buttons.

ffissore, 2015-05-26:
Menu size has been fixed with 17598368ac26a953a23634229b666e8b9a569ab6. The fix is already available with the hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-05-26 by ffissore