arduino/Arduino #3088: Serial Monitor is Not showing contents

sajanmittal786, 2015-05-06:
serial monitor of arduino IDE 1.6.3 is not showing any data of serial.print() command while other softwares like fritzing shows the data(i.e. no error in program). can any one help????

ffissore, 2015-05-06:
have you tried adding `Serial.flush` or using `Serial.println` instead?

sajanmittal786, 2015-05-06:
yes i tried it ... It gives me error  given below:
Arduino: 1.6.3 (Windows 7), Board: "Arduino Uno"

sketch_may06a.ino: In function 'void loop()':

sketch_may06a.ino:11:17: error: no matching function for call to 'HardwareSerial::flush(const char [2])'

sketch_may06a.ino:11:17: note: candidate is:

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:223:0,

                 from sketch_may06a.ino:2:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:118:18: note: virtual void HardwareSerial::flush()

     virtual void flush(void);

                  ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:118:18: note:   candidate expects 0 arguments, 1 provided

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

matthijskooijman, 2015-05-06:
You should be calling `Serial.flush()` without any arguments _after_ you did the `println()`, it doesn't replace the `println()`.

sajanmittal786, 2015-05-06:
i tried this too.. it didn't give any error but didn't work too. :(

sajanmittal786, 2015-05-06:
again serial monitor screen is empty..  
here is the code..

 float a;
void setup()
{
  Serial.begin(9600);
  
}

void loop()
{
Serial.println("-------------------");
Serial.flush();
a = millis();
Serial.print("a   ");
Serial.flush();
Serial.println(a);
Serial.flush();
Serial.println("-------------------");
Serial.flush();
delay(2000);
}

it scrolls the serial monitor but show nothing.

mschlenker, 2015-05-06:
First `millis()` does not return float, but unsigned long! Second: Is the baud rate matching?

sajanmittal, 2015-05-06:
hmmm yes baud rate is matched .... i have tested it by using other serial monitor like fritzing.. the output  properly displayed on that.. but serial monitor of arduino IDE  doesn't show any value
images  are attached for refernce...
[image]

[image]

sajanmittal, 2015-05-06:
here you can see baud rate is 9600 and value by  millis() is printed ... so does it matter where the value is float or unsigned long??? . if yes how??? if its about there length then the data is displayed through another monitor with same rate....

sajanmittal, 2015-05-06:
And sorry its their.. not there..

mschlenker, 2015-05-06:
The type thing does not break anything in such a simple sketch. Could you modify the sketch to stop printing after about 20 times and then try to select and copy from the serial monitor? Since the scrollbars are visible this might just be a problem of not properly updating the displayed output? 

Please also check with a non-aero windows theme. There are some combinations of graphics drivers and JVMs that tend to update the output too slowly thus sometimes not displaying output when transparency is used.

Lauszus, 2015-05-06:
@sajanmittal please notice the error in the Arduino IDE: "Error opening serial port 'COM3'. (Port busy)".

I suspect you have the serial port opened in Fritzing. You need to close the serial communication is Fritzing, as two program can not use the same serial port at the same time.

sajanmittal, 2015-05-06:
I think this is not the problem because  you can not   even open other serial monitor while other serial monitor is receiving data. it will show the error 

Error opening serial port 'COM3'. (Port busy)

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

since you can see in images that my serial monitor of Arduino IDE is opened, so it means my port is free at that time. these images are of different time. one serial monitor is closed while using other..

Avamander, 2015-05-26:
You have the port open somewhere else. Close all other programs and see if the issue persists.

agdl, 2015-10-02:
@sajanmittal any news about this?

closed 2015-11-26 by ffissore

akekapong78, 2018-05-22:
i found same ploblem, that my fault , and then i can fix ,  i forgot change chip processor on IDE.
[image]

JISexton, 2018-07-20:
akekapong78's suggestion has nothing to do with Arduino Due issue.

cmaglie, 2018-07-26:
This is not the an issue related to the Arduino IDE.
Please use the forum to get started or for assistance on your personal project.