arduino/Arduino #3090: Button example won't compile

Hansgur, 2015-05-06:
The provided example > digital > Button; won't compile, yet if one inserts just "Serial.begin(9600);" in setup it works fine. (as in "DigitalInputPullup").

What could the reason be? Bug?

mschlenker, 2015-05-06:
I tried under Linux/x64 with "Arduino AVR boards" in version 1.6.3, 1.6.5 and 1.6.6. Compiles everywhere. Which version do you use? Which platform?

matthijskooijman, 2015-05-06:
Compiles fine here as well. What error are you getting? What board do you have selected?

Hansgur, 2015-05-06:
Hi. Arduino Uno R3. 

>>Arduino: 1.7.2 (Windows XP), Board: "Arduino Uno"

collect2.exe: error: ld returned 5 exit status

Error compiling.
<<
  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

mschlenker, 2015-05-06:
Oh. Arduino 1.7.2 is over there: https://github.com/arduino-org/Arduino/issues

A long story:
http://blog.arduino.cc/2015/03/20/dear-arduino-community/
http://hackaday.com/2015/03/12/arduino-v-arduino-part-ii/

Hansgur, 2015-05-06:
Thanks I'll have a look what they say.

mschlenker, 2015-05-06:
Please take a look at the issue #2989 - this probably might happen with the .cc version 1.6.3 as well. XP is deprecated since more than one year now, so please do not expect anyone to be very eager to fix this.

Hansgur, 2015-05-06:
Hehe point taken  (XP)

mschlenker, 2015-05-07:
@ffissore Please merge and close #2989 and #3090

ffissore, 2015-05-08:
Sorry @mschlenker I'm not sure I caught that: what should I merge?

cmaglie, 2015-05-26:
Duplicate of #2989

closed 2015-05-26 by cmaglie