arduino/Arduino #3098: Improve new editor keyboard shortcuts, especially on mac

ffissore, 2015-05-07:
Starting from @damellis suggestions: https://github.com/arduino/Arduino/pull/3019#issuecomment-95699278

damellis, 2015-05-07:
Three in particular that would be nice to fix:
Alt-Backspace should delete the word preceding the cursor. (This works in 1.6.4).
Command-E should use the current selection for find. (This works in 1.6.4.) Also, Command-F should not use the current selection for find; it should use the previous find text. (This is wrong in 1.6.4, too.)
Command-Up should go to the start of the file; Command-Down to the bottom. (This doesn't work in 1.6.4 either.)

ffissore, 2015-05-13:
Alt-Backspace restored. Available in next hourly build

damellis, 2015-05-13:
Whoops, maybe I spoke too soon... it looks like this is deleting the word in front of the cursor, instead of behind it.

ffissore, 2015-05-13:
Whoops sorry. I coded that on purpose because I read your comment... Ehm...
Backwards

damellis, 2015-05-13:
I probably wasn't clear. Sorry about that!

ffissore, 2015-05-14:
Fixes pushed. From [your comment](https://github.com/arduino/Arduino/pull/3019#issuecomment-95699278) I miss only CMD+J: what does it do?

damellis, 2015-05-14:
Thanks again!

One minor nitpick: Cmd+E shouldn't open the Find dialog; it just sets the Find text (e.g. for use by subsequent Cmd+G commands).

In the current build, Cmd+J concatenates the next line onto the end of the current line. I find the equivalent behavior (J-key) useful in Vi, but here it seems out of place. Also probably a minor nitpick.

ffissore, 2015-05-15:
Uhm I guess we may leave CMD+E as it is (although it will look a little awkward, it will still implement the feature) and remove CMD+J on macosx (thus avoiding errors).
Closing, as the new editor should now behave as expected

closed 2015-05-15 by ffissore