arduino/Arduino #3103: Command-K doesn't work as shortcut for "Open Sketch Folder".

damellis, 2015-05-07:
In the latest version on GitHub (on Mac), Command-K doesn't open the folder containing the open sketch. It does highlight the Sketch menu as you'd expect on Mac, but the folder doesn't actually open. If you select "Show Sketch Folder" from the Sketch menu, the folder opens correctly. This seems to be true of both built-in examples and sketches in the sketchbook.

closed 2015-05-13 by ffissore

ffissore, 2015-05-13:
Fixed and available in hourly build

m-ri, 2016-09-25:
Caro,

Wow, ti  prego di dare un'occhiata a queste  belle cose che ho appena trovato, sono incredibili! Dai un'occhiata <http://giuseppa.mobiado.tw/estr>

Sinceramente tuo,marco ribero

m-ri, 2016-09-25:
Caro,

Ho delle  buone notizie per te, qualcosa che potrebbe piacerti molto, ti prego di  dare  un'occhiata <http://valter.mobiado.tw/fukc>

Cari saluti, marco ribero

commit 166c1aa7 (2015-05-13): New editor: restored CTRL+K (open sketch folder). Fixes #3103