arduino/Arduino #3106: Board Manager by URL

Testato, 2015-05-07:
I tested new function Boards Manager by just adding a url explained here:
http://blog.arduino.cc/2015/05/07/arduino-ide-1-6-4-released/

i tried the Adafruit URL posted by you here:
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

I installed from Board Manager the ESP8266 core and tried to compile.
On every code, included an empty setup+loop code, i receive:

    Arduino: 1.6.4 (Windows 7), Board: "Generic ESP8266 board"

C:\Users\win7\AppData\Roaming\Arduino15\packages\esp8266\tools\esptool\0.4.2/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -IC:\Users\win7\AppData\Roaming\Arduino15\packages\esp8266\tools\esptool\0.4.2/sdk/include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10604 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -IC:\Users\win7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.1\cores\esp8266 -IC:\Users\win7\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.1\variants\esp01 C:\Users\win7\AppData\Local\Temp\build5797500267345013060.tmp\sketch_may07a.cpp -o C:\Users\win7\AppData\Local\Temp\build5797500267345013060.tmp\sketch_may07a.cpp.o 

Error compiling.

ffissore, 2015-05-07:
You should ask Adafruit: boards manager "online help" links to https://forums.adafruit.com/

closed 2015-05-07 by ffissore

Testato, 2015-05-07:
Ok, this is the topic https://forums.adafruit.com/viewtopic.php?f=22&t=73280