arduino/Arduino #3107: [Web] Donate without credit card

mschlenker, 2015-05-07:
I want to donate €10 to support the development of Arduino without using a credit/debit card. Is there any possibility using a german bank account with IBAN/BIC for that purpose? Or if I lived in other parts of the world: PayPal or the like.

"Go, get away, contribute some pull requests!" would be some appropriate reaction to this issue while still not completely solving it.

mastrolinux, 2015-05-08:
Hello,

The costs of an international transfer and the EUR/USD conversion are much more than $10. This is why  we only accept CC right now. In the future we could accept Paypal but there is no ETA yet. 

I would like to thank you for the intention to contribute and indeed you are very welcome with your Pull Requests or improving the documentation.

mastrolinux, 2015-05-08:
I will close this issues since it is reported somewhere else.

closed 2015-05-08 by mastrolinux