arduino/Arduino #3110: Library Manager, doesn't update library

plinioseniore, 2015-05-08:
Hi,

I've pushed yesterday a new tag on the git repository for https://github.com/souliss/souliss/tags, but the Library Manager still see only v7.0-friariello.2 and doesn't recognize v7.0-friariello.3.

According to the documentation tags are scanned once per hour, could you please advice?

Thanks,
Dario.

plinioseniore, 2015-05-08:
Now the new tag has been recognized.

Has this been solved manually? Or simply the tags are refreshed at a slower rate than reported on the documentation?

Thanks.

ffissore, 2015-05-08:
Our internal server went down and we didn't notice

closed 2015-05-08 by ffissore

selengalp, 2019-01-09:
Hi,
My last release v1.3 on [this](https://github.com/sixfab/Sixfab_Arduino_CellularIoT_App_Shield/releases) doesn't show in arduino library manager. I tried to delete and release again but it didn't work. Is there any solution for me?