arduino/Arduino #3112: pulseIn doesn't functionality differs from API on Arduino.cc

rogerclarkmelbourne, 2015-05-09:
The code for pulseIn, on both AVR and SAM doesn't implement the API as described here
http://www.arduino.cc/en/Reference/PulseIn

in terms of the operation of the timeout value

https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/wiring_pulse.cpp

The spec states
"timeout (optional): the number of microseconds to wait for the pulse to start; default is one second (unsigned long) "

However all 3 while loops check for the timeout value (maxLoops) not just the first and second loops (or possibly not just the first loop)

Additionally, assuming that the spec is wrong, and that the timeout is the total time before the function will return.
Then some changes need to be made to both the first 2 waiting loops so that they take the same time to execute as the last loop (which actually measures the pulse width)

A simple work around for this is to declare 

```
volatile uint32_t dummyWidth = 0;
```

then in the waiting loops increment this dummyWidth value

Hence both the waiting loops have identical code to the actual timing loop and hence should take the same amount of time to execute

(Or just change the API and don't check maxLoops in the 2nd and 3rd while statements)

facchinm, 2015-05-09:
Hi @rogerclarkmelbourne , the `pulseIn` got a substantial rewrite to become gcc agnostic (see PR #2495 ) and the new meaning of timeout is expressed in 15df77a comments (and is handled by a single variable). SAM should include a similar rewrite btw, so thank you for reporting the problem!

facchinm, 2015-05-29:
fixed with #3112

closed 2015-05-29 by facchinm