arduino/Arduino #3114: gcc make

Coding-Badly, 2015-05-09:
Please include make.exe with the IDE distribution so the AVR bootloaders can more easily be rebuilt.

http://forum.arduino.cc/index.php?topic=64105.msg2224415#msg2224415

matthijskooijman, 2015-05-09:
This was previously reported as #2690, but that was closed for the reasons listed there. If you have something to add, please add comments that issue, I'm closing this one as a duplicate.

closed 2015-05-09 by matthijskooijman

Coding-Badly, 2015-05-09:
> If you have something to add...

It would take significantly less time for @ffissore to simply return make.exe to the distribution then it will take for me to convince @ffissore to change his mind.  So, this is the extent of my effort.

ffissore, 2015-05-12:
@Coding-Badly please request "make" inclusion at repo https://github.com/arduino/toolchain-avr/issues
Please consider providing its bash script: unfortunately, because of the hours it takes  to compile the toolchain, every modification takes days to complete