arduino/Arduino #3115: Cannot compile Example sketches. 1.6.4

jl-p, 2015-05-10:
Hi,

I upgraded to 1.6.4. and i was able to compile fastled library examples.
sometimes it hungs but most of the time it worked. 
but after a day i cannot upload any of the examples anymore.
I downloaded the zip version of 1.6.3. and the same codes with the same installed librarys works perfectly.

compile the code with 1.6.3. and save it again. open it in 1.6.4 and it get stuck again

output of the compiler

Bibliotheek FastLED in map: C:\Users\My Account\Documents\Arduino\libraries\FastLED (legacy) wordt gebruikt

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard -IC:\Users\My Account\Documents\Arduino\libraries\FastLED C:\Users\MYACCO~1\AppData\Local\Temp\build7734498561037075679.tmp\sketch_may10a.cpp -o C:\Users\MYACCO~1\AppData\Local\Temp\build7734498561037075679.tmp\sketch_may10a.cpp.o

jl-p, 2015-05-13:
I noticed my issue is also only when choosing something else then default as compiler warning settings

ffissore, 2015-05-25:
Fixed. Fix will be available with next hourly build. Thanks for the heads up!

closed 2015-05-25 by ffissore

z3t0, 2015-08-13:
Just to confirm, this fix is in 1.6.5?

ffissore, 2015-08-13:
Yes