arduino/Arduino #3120: ds1307rtc ReadTest sketch won't compile with 1.6.3 author:trainer111

trainer111, 2015-05-11:
Everything seems all right but the file will not compile.

ffissore, 2015-05-11:
Please paste error message

trainer111, 2015-05-11:
Hi Frederico,

Thank you for your prompt answer.

The sketch compiles ok with ver 1.5

The error message:
In file included from C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time\DateStrings.cpp:11:0:
C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time\DateStrings.cpp:41:22: error: variable 'monthNames_P' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 PGM_P monthNames_P[] PROGMEM = 
                      ^
C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time\DateStrings.cpp:58:20: error: variable 'dayNames_P' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 PGM_P dayNames_P[] PROGMEM = { dayStr0,dayStr1,dayStr2,dayStr3,dayStr4,dayStr5,dayStr6,dayStr7};
                    ^
C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time\DateStrings.cpp:59:24: error: variable 'dayShortNames_P' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 char dayShortNames_P[] PROGMEM = "ErrSunMonTueWedThrFriSat";
                        ^
Fel vid kompilering.

The last message means: error with compilation

I hope this will help finding the problem.

Best regards
Laszlo
Date: Mon, 11 May 2015 03:22:54 -0700

trainer111, 2015-05-11:
Hello again,

The file compiles ok with version 1.0.5-r2
not 1.5 as I stated earlier...
Sorry for the mistake.

Best regards
Laszlo

Date: Mon, 11 May 2015 03:22:54 -0700

Chris--A, 2015-05-11:
@trainer111 

The compiler has been upgraded since version 1.5.7.

It is a little more strict with certain things. PROGMEM data now is required to be marked const (and more accurately reflects the read-only aspect of PRGMEM data).

`char dayShortNames_P[] PROGMEM = "ErrSunMonTueWedThrFriSat";`

Now needs to be declared like:

`const char dayShortNames_P[] PROGMEM = "ErrSunMonTueWedThrFriSat";`

When using arrays of pointers to PROGMEM (array of strings) both the array and pointers need to be const.

`const char * const strList[] PROGMEM = { dayShortNames_P };`

cmaglie, 2015-05-11:
It seems that there is an old copy of the Time library in `C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time\` In this case I guess the simplest strategy is to update the "Time" library to the latest version. I suggest to use the library manager to do that.

trainer111, 2015-05-12:
Hello,

Thank you for your prompt reply.

I followed your suggestion (updated Time library with the library manager)
and got the following message with compilation:

C:\DOCUME~1\Laszlo\LOKALA~1\Temp\build5763987477002590569.tmp/core.a(wiring.c.o): In function `delay':
C:\Program\Arduino\hardware\arduino\avr\cores\arduino/wiring.c:114: undefined reference to `yield'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "Time.h"
 Used: C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time
 Not used: C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\arduino_367066
Fel vid kompilering.

"Fel vid kompilering" means: error at compilation...

Hope you can repeat the problem and let me know what's wrong...

Best regards
Laszlo

Date: Mon, 11 May 2015 07:27:46 -0700

trainer111, 2015-05-12:
Hello again,

Got the following message while compiling TimeRTC from examples:

C:\Program\Arduino\hardware\tools\avr\bin\avr-ar: unable to rename 'C:\DOCUME~1\Laszlo\LOKALA~1\Temp\build7364147201229355546.tmp/core.a'; reason: Permission denied
Multiple libraries were found for "Time.h"
 Used: C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time
 Not used: C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\arduino_367066
Fel vid kompilering.
C:\DOCUME~1\Laszlo\LOKALA~1\Temp\build7364147201229355546.tmp/core.a(wiring.c.o): In function `delay':
C:\Program\Arduino\hardware\arduino\avr\cores\arduino/wiring.c:114: undefined reference to `yield'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "Time.h"
 Used: C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time
 Not used: C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\arduino_367066
Fel vid kompilering.

This is after updating the Time library with the library manager.

Best regards
Laszlo

Date: Mon, 11 May 2015 07:27:46 -0700

ffissore, 2015-05-12:
@trainer111 do you have an antivirus? Can you double check if it's blocking/putting in quarantine files like ld.exe, objcopy.exe etc?

PaulStoffregen, 2015-05-12:
That message 'Multiple libraries were found for "Time.h"' is trying to tell you something!!!

trainer111, 2015-05-12:
Yes, one would think so!

I looked and there was only one instance of Time.h
It may be possible that during compilation the compiler generates some extra files
that are not seen at the end result...

Verrry confusing...

Date: Tue, 12 May 2015 08:41:13 -0700

PaulStoffregen, 2015-05-12:
@ffissore Do you remember an issue I reported some time ago, where Arduino doesn't detect some very wrong ways some users custom configure pathnames?

PaulStoffregen, 2015-05-12:
Ah, here it is, issue #2719.  But I'm not sure if that's the real problem.  Something strange does seem to be happening here.

If you look at the pathname, the conflicting libraries are in these locations:

        C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\Time

        C:\Documents and Settings\Laszlo\Mina dokument\Arduino\libraries\arduino_367066

Somehow, another copy of the Time library was put into a directory named "arduino_367066".  How could that be?  Was this name "arduino_367066" created automatically by the Arduino software?

trainer111, 2015-05-12:
Hello,

I'm just a simple user of Arduino sw.
I just load it down and and test the examples.

Haven't found issue #2719... Haven't looked yet

The timer example did work well on version 1.0.5 without a hassle.

I do not do anything out of the ordinary.
Anything gets done is done by the Arduino sw

I agree, something very fishy is going on...

Where does "arduino_367066" come from?

Definitely not from me to my knowledge!

Cheers
Laszlo

Date: Tue, 12 May 2015 08:57:45 -0700

trainer111, 2015-05-12:
Hi Paul,

Have checked issue #2719.

This seems to be the problem just like you described in the note.

I'm not sure what to do but I got some info to start on. Thank you!

I have version 1.0.5-r2 installed in my  "my documents" directory as well,
this could be causing the problems...
I need it until I verified that all my projects can be compiled and work
in the new IDE...

Too tired to do it now( had to take the kids to trainig)

Thanks again!

Cheers
Laszlo

Date: Tue, 12 May 2015 08:57:45 -0700

PaulStoffregen, 2015-05-12:
Well, there's 2 issues here.... what you need to do to get Arduino working, and when Arduino should do in the future to prevent anyone else from suffering these horribly confusing problems.

For you, probably the simplest thing would be to make a backup copy of any sketches and libraries you've written or modified.  Just delete everything from C:\Documents and Settings\Laszlo\Mina dokument\Arduino.  Then get a new copy of Arduino, and extract the ZIP file somewhere NOT inside Docuemnts / Arduino.

As for what Arduino should do to prevent confusing problems... well, that's more difficult.  Hopefully more attention will be given to issue #2719 and future versions of Arduino can do more to detect and prevent overlapping sketchbook and install directories.

trainer111, 2015-05-13:
Hi Paul,

This seems to be quite a mess so I thought I'd describe the sequence of events:

1: I was tempted by the new features offered by the new version like conditional compilation
    etc.
   I always expect hassle when a new version is introduced and I wasn't disappointed... 

2: Downloaded Ver 1.6.3 as one should. Automatically deleted the older version.
3: Tried to compile one of my sketches. Did not compile because of the timer isssue.
4: Tried to compile the example readRTC with the new version. Did not compile. Of course.
5: Downloaded and installed ver 1.0.5-r2.
6: Compiled example readRTC with 1.0.5. Worked
7: Compiled my sketch. Worked
8: Reported the problem. Was advised to install time library anew by library manager.
    Problem still there.
9: A new problem: Time example sketches do not compile with ver 1.6.3.

This is where we stand right now.

I think you should put in some warning messages that all libraries should be updated
as well when a new Andriod SW is installed.

I'm still not sure what I should do to get things right!

I hope this may give you some hints where the problem/problems is/are...

Best regards
Laszlo
Date: Tue, 12 May 2015 13:39:46 -0700

trainer111, 2015-05-21:
Hello again,

The problem seems to solved!
Thank you for your answers!

I moved version 1.0.5 to a new location and installed a new time(rtc) library and now compilation works.
The conditional compilation does not seem to work though.
I tried to use it for eliminating printing functions, but they still get compiled...
Should I enter a new issue/report?

closed 2015-05-26 by ffissore

Ian-PP, 2018-02-09:
Me happened the same, my solution has been so simple (sorry for the bad English I am Portuguese)

1st: install the latest version of the Time library 
         Link: [https://github.com/PaulStoffregen/Time/tree/master]