arduino/Arduino #3121: Latest Arduino corrupted sketch when upload failed

moebiusdx, 2015-05-12:
hi,

this is a deadly bug. i was uploading a sketch to Pro-Mini. the upload was hanging, so i unplugged the USB cable. after that, the sketch is completed corrupted. it turned into a binary file containing all NULs when viewed with editor.

has this been reported? anyway the sketch can be recovered.?

[image]

Chris--A, 2015-05-12:
Wow, that seems very dangerous indeed.

In your temporary folder where the sketch is copied and compiled should contain a file `{sketch-name}.cpp`. There should be a copy of the .ino in there. It will contain the IDE pre-processing, but still intact (hopefully).

On windows, a build directory looks like this: __*C:\Users\Chris\AppData\Local\Temp\build2502913623499128793.tmp*__

You could view the temp dir, then open the most recently modified buildXXX.tmp folder.

moebiusdx, 2015-05-12:
Hi,

 

I already checked the Temp folder. In it, contains the attached.

 

Also corrupted L

 

Thanks for the quick reply.

 

Regards,

Jon

ffissore, 2015-05-12:
Which version of the IDE are you using?

moebiusdx, 2015-05-12:
The latest from the site. I downloaded it 3 days ago.

ffissore, 2015-05-14:
Is it 1.6.4?

moebiusdx, 2015-05-14:
Yes.

moebiusdx, 2015-05-14:
Hi,

 

Attached is the actual corrupted sketch.

 

Thanks,

Jon

Avamander, 2015-05-18:
@moebiusdx You have to use an external file host.

moebiusdx, 2015-05-18:
Hi,

 

Sorry I did not understand your last response. Please elaborate.

 

Thanks,

Jon