arduino/Arduino #3124: IDE 1.6.4 Compiler warnings setup

pierrotm777, 2015-05-12:
I can compile only if i use the 'default' setting in setup ! (compiler stop with 'more' or 'all')  ...

matthijskooijman, 2015-05-12:
Can you show the full compiler output and errors you get? Verbose output is enabled in the preferences.

pierrotm777, 2015-05-12:
I don't understand , now all is ok with more and full !!!

luc-github, 2015-05-13:
I have same issue - it is stuck at the first compilation command if I change compiler warning:
Using library SPI in folder:
```
 C:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\libraries\SPI 

C:\Users\user\AppData\Roaming\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -Wall -Wextra -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10604 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/libsam -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/CMSIS/Include/ -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/Device/ATMEL/ -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\cores\arduino -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\variants\arduino_due_x -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\libraries\SPI C:\Users\user\AppData\Local\Temp\build4909961682515594001.tmp\Commands.cpp -o C:\Users\user\AppData\Local\Temp\build4909961682515594001.tmp\Commands.cpp.o 
```
but ok with default compilation warning:
```

Using library SPI in folder: C:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\libraries\SPI 

C:\Users\user\AppData\Roaming\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10604 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/libsam -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/CMSIS/Include/ -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/Device/ATMEL/ -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\cores\arduino -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\variants\arduino_due_x -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\libraries\SPI C:\Users\user\AppData\Local\Temp\build4909961682515594001.tmp\Commands.cpp -o C:\Users\user\AppData\Local\Temp\build4909961682515594001.tmp\Commands.cpp.o 
C:\Users\user\AppData\Roaming\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10604 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/libsam -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/CMSIS/Include/ -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\system/CMSIS/Device/ATMEL/ -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\cores\arduino -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\variants\arduino_due_x -IC:\Users\user\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.4\libraries\SPI C:\Users\user\AppData\Local\Temp\build4909961682515594001.tmp\Communication.cpp -o C:\Users\user\AppData\Local\Temp\build4909961682515594001.tmp\Communication.cpp.o 
[.......]
```

luc-github, 2015-05-13:
I am not sure if it can help to dig in but seems no issue with empty sketch, in this case all compiler warning modes are working

ffissore, 2015-05-13:
Can you share the sketch code? Put it on [gist please](https://gist.github.com/)

luc-github, 2015-05-13:
It is not a simple sketch - it is a complete project - https://github.com/repetier/Repetier-Firmware/tree/work092/src/ArduinoDUE/Repetier and same with my fork https://github.com/luc-github/Repetier-Firmware-0.92/tree/master/src/ArduinoDUE/Repetier

matthijskooijman, 2015-05-13:
Does it hang on the same file every time? If so, could you see if it hangs if you just use that single file (which will lead to link errors later, but that's ok).

luc-github, 2015-05-13:
I also can reproduce the with a empty sketch - with DUE as board and only with compiler warning set to  "All" it is stuck a USBCore.cpp not first file like in my project, all others compiler warning settings seems ok, if I use "uno" board no issue with empty sketch and all compilers warning settings

per1234, 2015-05-17:
I have the same problem. Under Arduino 1.6.4 or Hourly Build 2015/05/15 6:16 compiling any of the GSM examples with Compiler Warnings set to **More** or **All** freezes at the line:
```C:\Program Files (x86)\arduino-nightly\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\arduino-nightly\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\arduino-nightly\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\arduino-nightly\libraries\GSM\src C:\Program Files (x86)\arduino-nightly\libraries\GSM\src\GSM3ShieldV1.cpp -o C:\Users\per\AppData\Local\Temp\build1506494698685005532.tmp\GSM\GSM3ShieldV1.cpp.o ```
This happens with any Board. With Compiler Warnings **None** or **Default** it compiles fine. All other examples I have tried compile fine. I am using Windows 7 64bit.

pierrotm777, 2015-05-17:
Always this issue with an other sketch . With the 1.6.3 , when the compiler found an error , it stop . With the 1.6.4 version , it doesn't stop and stay on a .cpp

per1234, 2015-05-17:
@pierrotm777 have you tried to compile one of the GSM examples included with the Arduino IDE? Do you also have the same result of the Compiler Warnings setting  of More or All causing it to freeze during the compile.

pierrotm777, 2015-05-17:
Just compiled for a test the example 'MakeVoiceCall' and all is right :-)

per1234, 2015-05-17:
Does it compile for you with Compiler Warnings **More** or **All** mode using Arduino IDE 1.6.4?

pierrotm777, 2015-05-17:
It's OK with the More mode but not the All mode .

per1234, 2015-05-17:
Well at least we are having problems with the same file if different results. For me compiling GSM **MakeVoiceCall** with Compiler Warnings **More** freezes at the line:
```
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\libraries\GSM\src C:\Program Files (x86)\Arduino\libraries\GSM\src\GSM3ShieldV1CellManagement.cpp -o C:\Users\per\AppData\Local\Temp\build1147299745128299605.tmp\GSM\GSM3ShieldV1CellManagement.cpp.o 
```
And with Compiler Warnings **All** freezes at:
```
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\libraries\GSM\src C:\Users\per\AppData\Local\Temp\build1147299745128299605.tmp\MakeVoiceCall.cpp -o C:\Users\per\AppData\Local\Temp\build1147299745128299605.tmp\MakeVoiceCall.cpp.o 
```
I am also having the problem where an error in the code causes compile under 1.6.4 to freeze without displaying the error message regardless of the Compile Warnings setting but with 1.6.3 the compile fails and shows the error message as expected. I have not been able to narrow down a simple code example to demonstrate this yet so if you have a sketch that demonstrates this problem please post it. It seems like a separate but probably related issue.

per1234, 2015-05-17:
#3115 appears to report a similar issue.

per1234, 2015-05-24:
@ffisore I don't know what exactly is meant by the ***Board: Arduino Due*** tag you added, but just to be clear this problem occurs for me with any Board selected.

ffissore, 2015-05-25:
You're right. Fixed. Fix will be available with next hourly build. Thanks for the heads up!

closed 2015-05-25 by ffissore

per1234, 2015-05-25:
The fix solves this issue for me. Thank you!

commit 8a1e6c8a (2015-05-25): CollectStdOutExecutor and CollectStdOutStdErrExecutor were plain wrong, were losing data and were blocking compilation. Fixes #3124 and #3115