arduino/Arduino #3125: IDE 1.6.4 3rd party boards lost

pierrotm777, 2015-05-12:
Sorry for my english!

I try to add unofficial boards found on this link :
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

All is ok , but i can only see in the boards list the last json file used !
It's not possible to have a main file with all our son links ?

ffissore, 2015-05-12:
can you copy paste the content of the "Additional Boards Manager URLs" field in preferences?

pierrotm777, 2015-05-12:
I found found my error !
The good method is to add the url of the json files with a comma (,) between each url , like that !!!
boardsmanager.additional.urls=https://adafruit.github.io/arduino-board-index/package_adafruit_index.json,https://per1234.github.io/Ariadne-Bootloader/package_codebendercc_ariadne-bootloader_index.json,https://raw.githubusercontent.com/TKJElectronics/Balanduino/master/package_tkj_balanduino_index.json,https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json,https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json,https://raw.githubusercontent.com/carlosefr/atmega/master/package_carlosefr_atmega_index.json

Now , i can see all my board ! A good add for the version 1.6.5 would be the possibility to edit easily this option !

ffissore, 2015-05-12:
You can already: it's in File > Preferences. Just remember to separate each url with a comma

closed 2015-05-12 by ffissore