arduino/Arduino #3136: Add MagStripe library to the Library Manager

carlosefr, 2015-05-14:
I'd like to propose my MagStripe library for inclusion in the Library Manager:

    https://github.com/carlosefr/magstripelib

It supports decoding any of the standard three tracks (not simultaneously) from magnetic stripe cards, using a TTL card reader connected directly to an Arduino. The code has been stable for a few years.

carlosefr, 2015-05-14:
The repository has some extra files that probably can't be distributed with the IDE. The proper link is thus the latest release (which includes a .ZIP file with those extras stripped out):

https://github.com/carlosefr/magstripelib/releases/tag/v1.8.1a

ffissore, 2015-05-15:
@carlosefr what do you mean that these extra files can't be distributed with the IDE? Also remember that the only human activity involved in adding a library to the list is adding its repo url. A job fetches and tries to publish EVERY tag, not just the one you specified. See https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

carlosefr, 2015-05-15:
There were some PDFs in the repository that I collected from elsewhere for safekeeping. I've moved them to another branch and re-tagged the releases.

It should be OK now.

ffissore, 2015-05-18:
Added. Thanks!

closed 2015-05-18 by ffissore