arduino/Arduino #3137: Add PCD8544 library to the Library Manager

carlosefr, 2015-05-14:
I'd like to propose my PCD8544 library for inclusion in the Library Manager:

    https://github.com/carlosefr/pcd8544

This library is able to drive monochrome LCDs based on the Phillips PCD8544 or compatible chips. Unlike the Adafruit library already included, this doesn't support fancy graphic primitives, its main purpose is to use the least amount of resources possible. The code has been stable for a few years.

carlosefr, 2015-05-14:
The repository has some extra files that probably can't be distributed with the IDE. The proper link is thus the latest release (which includes a .ZIP file with those extras stripped out):

https://github.com/carlosefr/pcd8544/releases/tag/v1.4.1a

ffissore, 2015-05-15:
Blocked by #3136

carlosefr, 2015-05-15:
I made the same changes as for #3136, it should be OK now.

ffissore, 2015-05-18:
Added. Thanks!

closed 2015-05-18 by ffissore