arduino/Arduino #3140: Preferences window opens too small to use

richrad, 2015-05-14:
Like so:
[image]
I can fix it by trashing my `~/Library/Arduino/preferences.txt` file but I didn't figure out which line in particular is causing the problem. Here's a gist with the contents: [https://gist.github.com/richrad/47f2684cd1fd4f1f1aca](https://gist.github.com/richrad/47f2684cd1fd4f1f1aca)

I do have three or four versions of the IDE installed so I wouldn't be surprised if another version caused the problem.

ffissore, 2015-05-14:
We started coding a new preferences window, which should be corretly
displayed. More info later tomorrow

ffissore, 2015-05-15:
Fix will be available in next hourly build

commit 7816ee3f (2015-05-15): New Preferences window: renders fine on every OS and it's easier to adapt using NetBeans as visual editor. Fixes #3140
closed 2015-05-15 by ffissore