arduino/Arduino #3142: Editor crashes when program reaches certain size

dan-racke, 2015-05-14:
Hi, I have a problem. I am programming a Mega 2560. So far in the editor I only have around 2800 lines of code written. When I try to add more lines to the program and then compile I get the following error:

collect2.exe: error: ld returned 5 exit status
Error compiling.

This seems to happen around 2820 lines in the program.
What's up with this. I am barely scratching the memory available in the Mega, but it seems I cannot make a program big enough to fill it.

mschlenker, 2015-05-15:
Are you using XP? You might have run into https://github.com/arduino/Arduino/issues/2989

dan-racke, 2015-05-15:
I am using XP and I followed the various posts and loaded 1.0.6.. Yup, it fixed my problem, unfortunately, 1.6.4. did have a nice feature of allowing me to pre-initialize structures, and it would report the amount of memory used by the program. But, hey I can live with the loss of that as long as it lets me make large programs that will compile without errors :)

closed 2015-05-15 by dan-racke