arduino/Arduino #3145: No copyright header. License confusion.

mtomczew, 2015-05-15:
What is the license for file https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/cores/arduino/avr/pgmspace.h
?

It does not have any copyright header.
Is the licence LGPL like dtostrf.c file in the same directory?

It would be great if you could add appropriate header to decrease confusion how this file can be used in other software.

Additionally, writing "in general" in license.txt is not very clear. If part of code is licensed under GPL and the other under LGPL, specifying precise folders licensed differently is a good idea.

cmaglie, 2015-05-19:
This is something I added months ago, just after the Arduino Due was released.
I know that @PaulStoffregen has something very similar for the Teensy: I've looked before writing this one.

I'm going to add an LGPL license as for the rest of the core.
@PaulStoffregen is it ok for you?

PaulStoffregen, 2015-05-19:
Oh, looks like my copy of pgmspace.h ( https://github.com/PaulStoffregen/cores/blob/master/teensy3/avr/pgmspace.h ) lacks a license header too.

I use MIT license.  If the code came from me, please put the MIT license header on it.

cmaglie, 2015-05-19:
I really don't remember, it may be a mix of forum posts and your implementation. BTW I've added the MIT license, so no problem.

closed 2015-05-19 by cmaglie

mtomczew, 2015-05-20:
Thanks for help. License is appropriate for my usage.

commit 977ccc69 (2015-05-19): Added missing license on samd's pgmspace.h

Fixes #3145