arduino/Arduino #3146: OSCOKIRQ failed to assert on 1.6.2/3/4

r2DoesInc, 2015-05-15:
Using the USB Host 2.0 library BlinkLED example - and anything that uses this library in general - fails on 1.6.2, 1.6.3, 1.6.4. After reverting to 1.6.1, all works as expected.

Lauszus, 2015-05-15:
@r2DoesInc please open up an issue at the repository: https://github.com/felis/USB_Host_Shield_2.0/issues and I will be happy to help you out. As far as I understand from your issue, this is not really Arduino related, but is related to the library it self.

Btw please provide some more information i.e. what example are you talking about? I guess this one: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/adk/ArduinoBlinkLED/ArduinoBlinkLED.ino?

Also what do you mean by fail? Doesn't it compile or what? I just tried to compile it and everything seems fine.

Lauszus, 2015-05-15:
Also please specify what board and shield you are using.

ffissore, 2015-05-15:
I was just about to comment the same thing. Closing as invalid

closed 2015-05-15 by ffissore

r2DoesInc, 2015-05-15:
Apologies, I have recreated the issue with more info in the correct place.