arduino/Arduino #3168: conhost.exe

g4fkk, 2015-05-16:
Running the Arduino IDE v1.6.4 appears to cause conhost.exe to start and stop continuously.  This slows down the PC considerably.  The previous IDE I used (v1.0.4), seems not to start conhost at all.

The PC is running Windows 7 Home Premium, 64bit.

Is this a bug or a feature?

Regards - Martin

ffissore, 2015-05-18:
How do you know the IDE is restarting such process?

g4fkk, 2015-05-18:
I noticed Windows was running slowly and using a lot of memory so I checked TaskManager.  I can see conhost starting and stopping continuously in the 'processes' tab.  When I close the IDE, conhost stops trying to start and the PC goes back to normal so I'm guessing the IDE is triggering the effect one way or another.  Many thanks for your interest.
Martin

TheBlueTroll, 2015-05-19:
conhost.exe is the "Console Windows Host" (source "http://www.howtogeek.com/howto/4996/what-is-conhost.exe-and-why-is-it-running/")

its not surprising that it is loaded when the ide is opened, since the ide uses CLI (or Command Line Interface) programs such as avrdude or most likely with the ide opening, "listComPorts.exe". listComPorts.exe is a program that lists comm ports on windows, and is used to figure out how to communicate with the arduino. (image shows its CLI "http://todbot.com/blog/2012/03/02/listcomports-windows-command-line-tool-for-usb-to-serial/")

i imagine conhost keeps opening so that the ide can detect new serial ports. so yeah, its gonna keep opening, and killing it isnt gonna help.

now that we know what conhost is, and how the ide is using it, how do you fix it. in short, you can not. its not really a issue, as conhost appearing is nessary for the correct operation of the ide, and is the expected action of opening the ide. 

as for 1.0.4, the ide used the rxtx libary to do comm port detection, which was kinda buggy and much of a hack, and since rxtx was java, it didn’t need to be run as a CLI, so No CLI, No conhost.exe would spawn in 1.0.4

and at some point rxtx was probably dropped for listComPorts.exe.

in conclusion, conhost isnt the issue, at all. my suggestion is try a disk defrag ("http://www.pcadvisor.co.uk/how-to/windows/3493358/how-defrag-in-windows-7-windows-8/"), check what else is running, clear temp folder/ cache ("http://www.pcadvisor.co.uk/how-to/software/3273110/how-to-clean-up-the-temp-folder-in-windows/") and then try again.

ffissore, 2015-05-19:
I confirm what @TheBlueTroll says. On Windows the IDE will spawn listComPorts every 3 seconds: this is necessary to keep the IDE responsive with what the user does with her pc, plugging and unplugging arduinos in this case. I'm likely to close this issue as invalid: we can improve listComPorts but we cannot change how Windows handles programs run by other programs

g4fkk, 2015-05-21:
OK - well thanks for checking it out.  I do run other progs which use conhost but I guess they don't run it every 3 seconds.  It's a shame as it's making the processor in my PC run hot and the fans run up to full tilt, generating a cacophonous din.  The whole thing becomes slow and reminds me of the bad old Windoze 98 days :)  IDE 1.0.4 using the rxtx library works fine for me so I'll stick with it for the time being although I do like some of the new stuff on 1.6.4

Thanks again - Martin

closed 2015-05-21 by ffissore