arduino/Arduino #3169: Long Boot Time

Dokta42, 2015-05-16:
Suddenly the IDE is taking over four minutes to boot on WIN7 64b system.  Opening logo doesn't extinguish when it does boot, remaining on the screen.  Using version 1.6.4.

Chris--A, 2015-05-16:
On the first run of the IDE it unpacks the default AVR core and boards from a zip file. On slower machines this can take quite some time. If you closed the app while it was extracting, the process may have to be restarted next run.

Dokta42, 2015-05-16:
Thanks for your prompt response Christopher, however, I have been using IDE V1.0.6 for some time now, but the day before yesterday it suddenly took a very long time to load and failed repeatedly after about ten minutes (nothing has changed on my system).  I then upgraded to V1.6.4 And the problem continued, with a boot time of more than four minutes repeatedly.  I must have booted five or six times (uninterrupted) as I know the first boot is a little slow, but it was the same every time.  The intro logo was especially annoying as even after the IDE completed booting, the logo remained on the screen and in front of everything else.  

 

My system is not ‘slow’ at 3.6GHz x 6Core Intel processor, I would expect even the first boot to take considerably less than four minutes.

 

The problem continues…

 

Thanks

 

Dokta42

ffissore, 2015-05-18:
@Dokta42 are you saying you used 1.6.4 with no problems and then, one day, it started taking a long time to start?

Dokta42, 2015-05-18:
No, I was using 1.0.6 when it started slowing (I think, because I am away from my main computer – refer to my first email)...the problem continued AFTER upgrading to 1.6.4.  However, today, on the second boot things returned to normal.  The only difference was a Windows ‘Update’ between boots – could Windows be the problem?  All working okay now.

 

Regards,

 

Graeme

Chris--A, 2015-05-18:
Do you use GitHub often? 
On my PC, using GitHub and avast anti virus causes a race condition in ssh_agent. It slows down my PC to point I sometimes have to restart, occasionally forcefully. Windows updates waiting seems to be a factor in this starting, as it disappears as soon as I have updated everything.

ffissore, 2015-05-18:
It looks like an issue with windows then, not directly related to the IDE

closed 2015-05-21 by ffissore