arduino/Arduino #3170: SPI library not found

Varbin, 2015-05-16:
I wanted to use a SD-Card with my Arduino Uno + Ethernet Shield (code from http://www.arduino.cc/en/Tutorial/DumpFile) but I got following error:

```
/home/simon/builds/arduino-1.6.4/libraries/SD/src/utility/Sd2Card.cpp:26:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>
                 ^
compilation terminated.
Fehler beim Kompilieren.
```

Chris--A, 2015-05-16:
It needs to be included in the sketch also (this will hopefully be fixed soon, part a feature of the IDE preprocessing).

Include SPI.h and SD.h into your sketch.

agdl, 2015-05-21:
@Varbin @Chris--A I linked the example to dump the code from Git. Now it is correct.

closed 2015-05-21 by agdl