arduino/Arduino #3171: "Export Compiled Binary" errors

Isaac96, 2015-05-16:
I just downloaded the "Arduino 1.6.5 Hourly Build 2015/05/15 06:16" Arduino software, and tried out the new "Export Compiled Binary" option without doing anything else. I got the first line highlighted and a whole lot of errors:

C:\Users\Isaac\AppData\Local\Temp\build1926227438616564483.tmp/sketch_may16a.cpp.elf C:\Users\Isaac\AppData\Local\Temp\build1926227438616564483.tmp/sketch_may16a.cpp.hex 
processing.app.debug.RunnerException
	at processing.app.debug.Compiler.saveHex(Compiler.java:1172)
	at processing.app.debug.Compiler.compile(Compiler.java:437)
	at processing.app.debug.Compiler.build(Compiler.java:117)
	at processing.app.Sketch.build(Sketch.java:1162)
	at processing.app.Sketch.build(Sketch.java:1137)
	at processing.app.Editor$BuildHandler.run(Editor.java:2057)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.helpers.PreferencesMapException: recipe.output.tmp_file
	at processing.app.helpers.PreferencesMap.getOrExcept(PreferencesMap.java:268)
	at processing.app.debug.Compiler.saveHex(Compiler.java:1162)
	... 6 more

ffissore, 2015-05-18:
That's probably because you have installed a avr core with boards manager: that core wins over the bundled one but it misses export recipes. I'll add checks for avoiding such issue. Thanks for notifying

ffissore, 2015-05-18:
Fixed. Fix will be available with next hourly build

commit 4c3de266 (2015-05-18): Don't export sketch if the underlying core does not support it. Fixes #3171
closed 2015-05-18 by ffissore