arduino/Arduino #3174: [Ubuntu 1.6.4] Xorg crashes when scrolling down the boards manager

NicoHood, 2015-05-17:
I have several boards installed within my sketchbook/hardware folder. Tried with my HoodLoader2, HID-Project and the sparkfun folder.

Thats a lot of boards, so you have to scroll down to reach the last entry. However if you try to scroll down the whole X session crashes and you have to login again. Every program like browser crashes as well.

Tried with lastest standard Ubuntu 64 bit, IDE 1.6.4 and java7. I did not test any older release yet.

ffissore, 2015-05-18:
Does it happen with latest nightly?

NicoHood, 2015-05-18:
Yes, problem still there. X_x

ffissore, 2015-05-20:
Confirmed. Veeery annoying.

ffissore, 2015-05-21:
Fixed. Fix will be available with next hourly build

ffissore, 2015-05-21:
Nope, bug is still there

ffissore, 2015-05-21:
Various attempts to solve the issue failed. However, when running the IDE with latest java 8, the issues did not show up. I then suggest you to install latest java 8 following [this tutorial for ubuntu](http://ubuntuhandbook.org/index.php/2015/01/install-openjdk-8-ubuntu-14-04-12-04-lts/) or downloading it directly from [oracle website](http://www.oracle.com/technetwork/java/javase/downloads/index.html) and changing the `java` command [here](https://github.com/arduino/Arduino/blob/master/build/linux/dist/arduino#L31) specifying the full path (mine is `/opt/jvm/jdk1.8.0/bin/java`)

closed 2015-05-21 by ffissore

NicoHood, 2015-05-25:
The problem is still there.
Telling the people to install java8 and pretend the bug is fixed is not a solution. I suggest two things:

Add a note to the installing instructions/download section that you need java 8 (not tested for me though)
Or open the issue and give away some cash if someone else is able to fix it.

Its a critical bug, all programs crash, all data is gone because of this fucking IDE. If someone accidentally gets this bug, this will drive him crazy. Especially when you havent saved your code/docs.

I noticed some changes in the nightly (of course). It would be nice if you could pin the Arduino IDE to the Ubuntu launcher as well. I've created a shortcut myself but it will open a 2nd Icon on the launcher.

ffissore, 2015-05-25:
Please avoid using harsh words. If the problem lies in how the jvm handles graphics, we can't do much but ask to use an updated jvm.

NicoHood, 2015-05-25:
It wasnt ment to be harsh. I just remembered how I was surprised when my X Server closed the first time.

Do I have to uninstall java 7 first?
I will try that and let you know if it works. You should add those notes somewhere then.

ffissore, 2015-06-10:
Linux version will bundle the JVM, thus solving this issue

NicoHood, 2015-07-16:
I had this program today again.
I had 1.6.6 open and also started 1.6.5 at the same time. Both via commandline, but sadly didnt log the output.

1.6.5 opened with a few windows, I closed them and then the whole screen died. Music went off after a while and I had to relogin with a new clean session. Same as the issue above.
As far as I can see java8 is already bundled in 1.6.5? I have at least not manually installed java.
Using Elementary os now, based on ubuntu 14.04

Here is my xsession-errors.old. Thats all I have right now. This could be a new problem, but related to this issue. I had it only once so I did not open a new issue for now.

I think the crash starts after the Rythmbox Robin Schulz logs:
```
Script for ibus started at run_im.
Script for auto started at run_im.
Script for default started at run_im.
Script for ibus started at run_im.
Script for auto started at run_im.
Script for default started at run_im.

** (gnome-session:1688): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-8Pse0Q6lVX: Connection refused

** (gnome-session-check-accelerated:1820): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-8Pse0Q6lVX: Connection refused
gnome-session[1688]: WARNING: Error while executing session-migration: Failed to execute child process "session-migration" (No such file or directory)
gnome-session[1688]: WARNING: Could not parse desktop file user-dirs-update-gtk.desktop or it references a not found TryExec binary
gnome-session[1688]: WARNING: Could not parse desktop file user-dirs-update-gtk-pantheon.desktop or it references a not found TryExec binary

** (gnome-settings-daemon:1828): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-8Pse0Q6lVX: Connection refused

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): keyboard-plugin-WARNING **: Unhandled settings change, key 'input-sources-switcher'

(gnome-settings-daemon:1828): keyboard-plugin-WARNING **: Unhandled settings change, key 'bell-custom-file'

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.

(gnome-settings-daemon:1828): media-keys-plugin-WARNING **: Grab failed for some keys, another application may already have access the them.
[... 693 lines cut ...]
g_dbus_connection_real_closed: Remote peer vanished with error: Underlying GIOStream returned 0 bytes on an async read (g-io-error-quark, 0). Exiting.
g_dbus_connection_real_closed: Remote peer vanished with error: Underlying GIOStream returned 0 bytes on an async read (g-io-error-quark, 0). Exiting.

(process:1992): indicator-application-service-CRITICAL **: Unable to send ApplicationRemoved signal: The connection is closed

(process:1992): indicator-application-service-WARNING **: Name Lost
[... cut ...]
g_dbus_connection_real_closed: Remote peer vanished with error: Underlying GIOStream returned 0 bytes on an async read (g-io-error-quark, 0). Exiting.

(process:1986): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Invalid MIT-MAGIC-COOKIE-1 keyInvalid MIT-MAGIC-COOKIE-1 key
** (plank:19196): WARNING **: Could not open X display
Invalid MIT-MAGIC-COOKIE-1 keyInvalid MIT-MAGIC-COOKIE-1 keyUnable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

(plank:19196): Gtk-WARNING **: cannot open display: :0
```

NicoHood, 2015-07-16:
/var/log/Xorg.o.log.old (last lines)
```
[ 17261.960] (II) AIGLX: Resuming AIGLX clients after VT switch
[ 17261.960] (II) intel(0): switch to mode 1600x900@60.0 on eDP1 using pipe 0, position (0, 0), rotation normal, reflection none
[ 17262.772] (II) intel(0): switch to mode 1280x1024@60.0 on VGA1 using pipe 1, position (1600, 0), rotation normal, reflection none
[ 17262.821] (--) synaptics: SynPS/2 Synaptics TouchPad: touchpad found
[ 17262.885] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.907] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.914] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.921] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.931] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.941] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17262.951] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.964] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.977] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17262.993] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17263.005] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17263.022] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.022] (II) XKB: reuse xkmfile /var/lib/xkb/server-81907AB388B478D711E6423DD8B0048239BE0CB6.xkm
[ 17263.073] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.114] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.155] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.195] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.237] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.283] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.326] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.370] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17263.409] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 17328.312] (II) XKB: reuse xkmfile /var/lib/xkb/server-F368D20DF3963C9041B5140771273B3C0AD33E46.xkm
[ 18630.621] (EE) 
[ 18630.621] (EE) Backtrace:
[ 18630.622] (EE) 0: /usr/bin/X (xorg_backtrace+0x48) [0x7fa74c0c0ed8]
[ 18630.622] (EE) 1: /usr/bin/X (0x7fa74bf1c000+0x1a8f19) [0x7fa74c0c4f19]
[ 18630.622] (EE) 2: /lib/x86_64-linux-gnu/libc.so.6 (0x7fa74a09c000+0x36d40) [0x7fa74a0d2d40]
[ 18630.622] (EE) 3: /usr/lib/xorg/modules/drivers/intel_drv.so (0x7fa745ea2000+0x463ca) [0x7fa745ee83ca]
[ 18630.622] (EE) 4: /usr/lib/xorg/modules/drivers/intel_drv.so (0x7fa745ea2000+0x4f475) [0x7fa745ef1475]
[ 18630.622] (EE) 5: /usr/bin/X (0x7fa74bf1c000+0x131288) [0x7fa74c04d288]
[ 18630.622] (EE) 6: /usr/bin/X (0x7fa74bf1c000+0x4ff0f) [0x7fa74bf6bf0f]
[ 18630.622] (EE) 7: /usr/bin/X (0x7fa74bf1c000+0x5369e) [0x7fa74bf6f69e]
[ 18630.622] (EE) 8: /usr/bin/X (0x7fa74bf1c000+0x574ea) [0x7fa74bf734ea]
[ 18630.622] (EE) 9: /lib/x86_64-linux-gnu/libc.so.6 (__libc_start_main+0xf5) [0x7fa74a0bdec5]
[ 18630.622] (EE) 10: /usr/bin/X (0x7fa74bf1c000+0x42a5e) [0x7fa74bf5ea5e]
[ 18630.622] (EE) 
[ 18630.622] (EE) Segmentation fault at address 0x7ffd04d58b00
[ 18630.623] (EE) 
Fatal server error:
[ 18630.623] (EE) Caught signal 11 (Segmentation fault). Server aborting
[ 18630.623] (EE) 
[ 18630.623] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[ 18630.623] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[ 18630.623] (EE) 
[ 18630.623] (II) AIGLX: Suspending AIGLX clients for VT switch
[ 18631.838] (EE) Server terminated with error (1). Closing log file.
```

commit 17598368 (2015-05-21): Updated MenuScroller with latest from https://tips4java.wordpress.com/2009/02/01/menu-scroller/
Reviewed its getMaximumItems method (now getMaximumDrawableMenuItems) so that it returns a 1/3rd less the max number, avoiding a bad crash on linuxes. Fixes #3174