arduino/Arduino #3176: IDE used to open with last sketch

Mokolea, 2015-05-17:
Arduino IDE 1.6.4 on MAC OS X does not open with last used sketch anymore!

In `preferences.txt` there are tags like:
  - `last.sketch.count=`
  - `last.sketch0.location=`
  - `last.sketch0.path=`

These were used in previous IDE

Chris--A, 2015-05-17:
On building the latest source (for 1.6.5) it opened with my last viewed sketch.

Mokolea, 2015-05-17:
With the currently latest available nightly build (Mac OS X, 2015-05-15) the problem is still there.

Mokolea, 2015-05-20:
I think the problem is only in writing/updating those mentioned key value pairs.
They are not updated if you quit the Arduino IDE.

If you however edit them manually (set `last.sketch.count=1`), and then startup the IDE, it opens as usual with the last sketch.

ffissore, 2015-05-25:
This is fixed with cc4af81ac54313da03b0733cc754eb1f3d040374 and already available with the nightly build. If you close the IDE with File > Quit, it will store the currenly opened sketches and reopen them the next time. If you close each window separately, those closed sketches won't get reopened

closed 2015-05-25 by ffissore

Mokolea, 2015-05-26:
I tried the nightly build, it works
Thanks

MAnand59, 2016-10-18:
hey i have the same problem guys....whenever i close arduino ide after working on a sketch and again when i open the ide it popup with the last sketch i wrote by default...

cmaglie, 2016-10-18:
> whenever i close arduino ide after working on a sketch and again when i open the ide it popup with the last sketch i wrote by default...

This is the expected behaviour

atomkey, 2017-02-01:
Yeah - well how do you stop it? If I want it to load a file I will tell it. I expect the IDE to open blank.

atomkey, 2017-02-01:
Fixed. Do this: "C:\Program Files (x86)\Arduino\arduino.exe" C:\Users\Atomkey\Documents\Arduino\test\falken\falken.ino

In the properties (assuming you are running windows) of the target.

I created a default program called falken.ino as follows in my test folder where I usually start to prototype:

// Greetings Professor Falken

void setup() 
{

}

void loop() 
{

}

** No when I launch the IDE it always brings up this to start with. Make it a blank file if you like, but that's my fix for now until someone adds a checkbox in the preferences of the IDE 
[x] load last sketch on start

-cheers-

-----------------------

And why do people put the begin braces at the end of the line now-a-days?
 loop() {
}

instead of this:
loop()
{

}

Everywhere I see code this is going on and it looks terrible.

atomkey, 2017-02-01:
Okay - that kinda worked. The stupid IDE saves my sketch every single time I compile. So don't forget to remove that checkbox in the preferences. And I went ahead and made my falken.ino default read-only just to make sure I am forced to save it to a new distinct filename. The IDE will STILL verify and upload with no complaints about it being read-only. But upon closing the IDE it will ask you to save. If you click on save it gets upset about the file being read-only and produces a save window for your approval.