arduino/Arduino #3178: 1.6.4 crashes when run on Raspberry pi 2, raspbian wheezy

diemkae, 2015-05-17:
Running rapsberry Pi 2 with raspbian/wheezy:

   Linux raspi2 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux

Want to upgrade distro version of Arduino ide (1.0.1) to 1.6.4 but after downloading and "installing" (in /usr/local/bin/arduino/arduino-1.6.4) starting the IDE crashes with the (final) message:

/usr/local/bin/arduino/arduino-1.6.4/lib/libastylej.so: /usr/local/bin/arduino/arduino-1.6.4/lib/libastylej.so: cannot open shared object file: No such file or directory (Possible cause: can't load IA 32-bit .so on a ARM-bit platform)
Cannot load native library /usr/local/bin/arduino/arduino-1.6.4/lib/libastylej.so
The program has terminated!

diemkae, 2015-05-17:
BTW, (obviously) because it's and Intel binary:

    dmk@raspi2 ~ $ file /usr/local/bin/arduino/arduino-1.6.4/lib/libastylej.so
/usr/local/bin/arduino/arduino-1.6.4/lib/libastylej.so: ELF 32-bit LSB shared object, Intel 80386, version 1 (SYSV), dynamically linked, BuildID[sha1]=0xa916401a36d8e650cf88460f9738c485bc62e18b, stripped

diemkae, 2015-05-17:
That is not a "solution" ... it is just the cause of this problem  ;-)

mschlenker, 2015-05-18:
You'd better change the subject of this issue to "Feature request: Offer IDE packages for Raspbian Wheezy". Obviously just Linux on x86/x86-64 is supported. Even if you manage to run the IDE itself (by making minor changes to themeing) the toolchain and avrdude will fail.

diemkae, 2015-05-18:
Perhaps so, but Arduino 1.0.1 runs just fine on Raspbian/wheezy on the
Raspberry Pi (maybe because the Element14 folks worked on it a bit  ;-)

 "Liberals claim to want to give a hearing to other views, but then are
shocked and offended to discover that there are other views."

William F. Buckley, Jr.

ffissore, 2015-05-18:
@diemkae the ide uses stuff (executables and libraries) and most of them are available with apt-get. Feel free to hack on whatever needs to be hacked for the current IDE to run on raspberry pi. For instance, we build astyle binaries using this bash script https://github.com/arduino/astyle. Maybe pinging Element14 may jump in and port 1.6 to pi

mschlenker, 2015-05-18:
Yes, the version supplied by the Raspbian Repos is indeed built for ARMv6. Building the IDE itself should not be a big issue following the notes on building for Linux. Afterwards some configuration work has to be done to install newer versions of avr-gcc and avrdude (Jessie repos?) and change the config to point to those system installed binaries.

Having official binaries for the RPi would be nice of course.

ShorTie8, 2015-05-26:
@mschlenker  "and change the config to point to those system installed binaries."

Got any clues on how to do this easily ??

Up till 1.6.2 you could easily make symlinks that worked fine.
1.6.3 with the addition of the platform rewrite I had to hard code them.
1.6.4 with the addition of the signature checks, I have not been able to get it to run.

Any help would be most appreciated.

Thankz
Have A Great Day

sandeepmistry, 2016-03-09:
Marking this as a duplicate of #3334, there are builds available to try out in #4517.

closed 2016-03-09 by sandeepmistry