arduino/Arduino #3179: SerialEvent Tutorial BUG

GitFinn, 2015-05-17:
I have tried the Serial Event example in both 1.6.3 and 1.6.4
and in both cases the function serialEvent() failed.
In operation, it is expected that the last character is '\ n'
This just not happen.
In the "send" number two the first character is  '\ n' but the string does not end with '\n' as expected.

agdl, 2015-05-21:
@GitFinn I made a PR because I think that the problem is simply a missing call to the function. 
Here the PR.
https://github.com/arduino/Arduino/pull/3203

GitFinn, 2015-05-21:
The proposed change does not help.
Download the original program example and make the following changes.
void serialEvent() {
   while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '9') {  // <<<<<<<<<<<<<<<<<<<<<<<<<<<< end of string char is 9
      stringComplete = true;
    }
  }
  while (Serial.available()) { //<<<<<<<<<<<<<<<<<<<<<<< This ensures that the buffer is emptied
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    inputString += inChar;
    }

Change the main loop to:
void loop() {
  // print the string when a newline arrives:
  if (stringComplete) {
    Serial.println(inputString);
    Serial.print("Laengden af strengen  ");
    Serial.println(inputString.length());
    int x = inputString.length();
    int y = 0;
    while ( y < x ){
      Serial.print(y);
      Serial.print(" ");
      Serial.print(inputString[y]);
      Serial.print(" ");
      Serial.println(byte(inputString[y]));
      y++;
      }

    // clear the string:
    inputString = "";
    stringComplete = false;
  }
}

Programme an Arduino and
open the Seriel Monitor and enter            asd9     press Enter
You receive:
Laengden af strengen  4
0 a 97
1 s 115
2 d 100
3 9 57

Now Enter the following                         asd9    press Enter
Laengden af strengen  5
0  13                   <<<<<<<  This is wrong! 
1 a 97
2 s 115
3 d 100
4 9 57

The character 13 should be sent when you pressed Enter
and not as the first character in the subsequent string.
When you press Enter the string must be terminated with code 13
Code 13 is loaded into the buffer but not sent.
The fault may in Serial Monitor code.

Sincerely, Finn Johansen

GitFinn, 2015-05-21:
I'm sorry to have disturbed you!

I was not aware of that the menu at the bottom of SerialMonitor also affected the transmission of data.
I have not been able to find this documented anywhere.
In my 40 years of programming, I always send CR as terminator. I never dreamed that it could be turned off.

The example works fin as long as there is selected "Both NL & CR" or "Newline".

Now the old man learned something new again ;-)

Thanks for the fantastic toys - Arduino

closed 2015-05-22 by cmaglie