arduino/Arduino #3182: goto

Dene48, 2015-05-18:
All other Control Structures (if, for, switch case, while, do..while, break, continue, and return) all and are recognized by the software goto is not

ffissore, 2015-05-18:
Can you elaborate? What do you mean with *recognized* ?

Dene48, 2015-05-18:
Every other instruction under the label on your website “control
structures” turn red when you add them to your program goto does not. If
I use your goto example or a simple goto example I get the same error
“expected identifier before ‘;’ token “ what am I doing wrong? If you
wood please make up a simple stripped down program with a simple jump
that works on your computer and send it to me maybe I have corrupt
software but I don’t think it is I have reloaded it. Thank you for
returning my e-mail.

Please send me that program

Dennis Rinaldo

denerin@mailservice.ms

Links:

  1. https://github.com/arduino/Arduino/issues/3182#issuecomment-103113700
  
  denerin@mailservice.ms

Chris--A, 2015-05-18:
In version 1.6.4 of the IDE, `goto` is highlighted, and its always been available for use.

```arduino
void setup() {
  
  int i = 0;
  
  Serial.begin( 9600 );
LoopIt:
  Serial.println( i, DEC );
  delay( 500 );  
  i++;
  goto LoopIt;
}

void loop(){}
```

Dene48, 2015-05-19:
Thank you your code works, does not turn color on in 1.0.6. If I program
a goto to jump up in the code they all work. But the code in your goto
example and any that I try to jump down the code do not. Please see if
you can run your example code

Your Example

for(byte r = 0; r < 255; r++){

for(byte g = 255; g > -1; g--){

for(byte b = 0; b < 255; b++){

if (analogRead(0) > 250){ goto bailout;}

// more statements ...

}

}

}

bailout:

You have been more than tolerant, pull this one off I’ll go away
forever.

Dennis Rinaldo

> In version 1.6.4 of the IDE, goto is highlighted, and its always been
> available for use.

>
> void setup() {

  int i = ;

  Serial.begin( 9600 ); LoopIt: Serial.println( i, DEC ); delay( 500 );
  i++; goto LoopIt; }

> void loop(){} —

Links:

  1. https://github.com/arduino/Arduino/issues/3182#issuecomment-103227701
  
  denerin@mailservice.ms

Chris--A, 2015-05-19:
There is really no problem with goto. IMHO your errors are just a sign not to use it.

The code below may look nice if you like goto, however this unconditional control flow is silly in a high-level language. Jumping around should be limited to assembly.

```arduino
void setup() {
  Serial.begin( 9600 );
}

void loop(){

  static int i = 0;

  if( i % 2 ){
    goto DoOdds;
  }
  goto DoEvens;

DoOdds:
  Serial.print( "Number is odd: " );
  Serial.println( i );
  goto Fin;  
DoEvens:
  Serial.print( "Number is even: " );
  Serial.println( i );
Fin:
  ++i;
  delay( 500 );
  return;
}
```

ffissore, 2015-05-19:
Indeed `goto` is highlighted in 1.6.4, even if with the wrong color. Fixed in 11476df7f7c1f714e2d9295711fa7922a3d1b2a8 and available in next hourly build. As for goto not working at all, you should dig into your code for the actual reason: goto is proven working by @Chris--A example. BTW, I second his advice: don't use `goto`.

closed 2015-05-19 by ffissore

Dene48, 2015-05-19:
Thank you for your time, I have no interest in using goto just needed to
know why it wasn’t working.

Old retired engineers are like that!

Have a good day

Dennis Rinaldo

> There is really no problem with goto. IMHO your errors are just a sign
> not to use it.

> The code below may look nice if you like goto, however this
> unconditional control flow is silly in a high-level language. Jumping
> around should be limited to assembly.

>
> void setup() {
  Serial.begin( 9600 ); }

> void loop(){

  static int i = ;

  if( i % 2 ){ goto DoOdds; } goto DoEvens;

DoOdds: Serial.print( "Number is odd: " ); Serial.println( i ); goto
Fin; DoEvens: Serial.print( "Number is even: " ); Serial.println( i );
Fin: ++i; delay( 500 ); return; }

Links:

  1. https://github.com/arduino/Arduino/issues/3182#issuecomment-103286633
  
  denerin@mailservice.ms

commit 11476df7 (2015-05-19): Fixed goto keyword color. Fixes #3182