arduino/Arduino #3183: Port selection unavalible

PacksGamingHD, 2015-05-18:
If you want to make me go through the extra steps of choosing my board the least you could do is make the Port option selectable! 
[image]
Before it was bad enough that I had to go to File > Open > and choose the file instead of double clicking before!

Fix these bugs for crying out loud!

Mac OS X 10.9.5

ffissore, 2015-05-18:
Please don't cry and tell us which version of the IDE are you running

PacksGamingHD, 2015-05-18:
1.6.4

Chris--A, 2015-05-18:
If there is no Arduino connected to the PC via USB, or the PC does not recognize it, the port menu will be disabled.

- Try re-connecting (cheap/old cables do not help).
- Check your device is discovered by the PC (on windows there is a device manager, there would be something similar on a mac)
- If nothing, install/re-install the drivers (ARDUINO_PATH\drivers).

PacksGamingHD, 2015-05-19:
I'm on a mac and apparently I think it's because I don't actually have wifi (PCIE card broke or something) and now I use an adaptor.

ffissore, 2015-05-19:
Which boards exactly have you attached to your computer?

Why do you think your wifi adapter is the problem? Are you using a Yun? If
so, does attaching it via usb cable make it work?

PacksGamingHD, 2015-05-19:
No I'm using an Arduino Pro 3.3 v atmega 328 (TinyDuino)

The only two things connected were the USB cable for the Arduino and
the 802.11n WLAN Adapter

ffissore, 2015-05-19:
Can you say it's an issue with your tinyduino coupled with that wlan
adapter then?

PacksGamingHD, 2015-05-20:
I have no idea. The only port at the minute that works is usbmodem1411 which is for the Arduino Unos

Chris--A, 2015-05-20:
Try your other USB ports.

On my old PC, only the main ports on the back of the tower worked with the Arduino.
Using the secondary extension at the front (add on panel in case) does not work.

ffissore, 2015-05-20:
This starts looking like a problem on your computer, not with the IDE (proof is the UNO is showing up)

PacksGamingHD, 2015-05-20:
Just connected to Ethernet and all the ports are working

closed 2015-05-20 by PacksGamingHD

PacksGamingHD, 2015-05-21:
I have two USB ports. My WLAN adaptor only works in one and the other is for the Arduino