arduino/Arduino #3184: Serial.println behaviour change between IDE 1.0.5 and 1.6.4

MJP007, 2015-05-18:
I have a sketch used to test serial comms between XBee modules.  In IDE 1.0.5 it works fine, in 1.6.4 it appears to be writing out extra characters.

The 1.0.5 behaviour (correct) is
Send using TeraTerm
^1234$
Reply
^X---$
The 1.6.4 behaviour (wrong) is
Send using TeraTerm
^1234$
Reply
^X---$ Bad Length 

The text "Bad Length" is part of an error message that appears in the code but has not been sent as part of the reply. 

The sketch is below.  Please advise.  Thanks

/*
 Test TX RX buffering with sleep.
 Does sleep stop the RX buffer working
 */
int led_pin = 13;
int RXBuffSize = 5;
char IncomingBytes[7];
char IncomingByte;
char OutgoingBytes[6]= {
  '^','X','-','-','-','$'};

#define VERSION "1.00a0"

void setup() {

  Serial.begin(9600);
}
void loop() {
  byte CharsRead;
  if (Serial.available() > 0 ) {
    //read buffer untill cr is hit
    delay(10); //Give the trailing bytes time to arrive
    memset(IncomingBytes,0,sizeof(IncomingBytes)); //Fill memory with zeroes
    CharsRead = Serial.readBytesUntil('\n', IncomingBytes, sizeof(IncomingBytes));
    if (CharsRead != sizeof(IncomingBytes)) {
      Serial.print (CharsRead,DEC);
      Serial.println (" Bad Length");
    }
    else if (IncomingBytes[6] != 13) { // same as \n
      Serial.print (IncomingBytes[6],DEC);
      Serial.println (" Missing CR");
    }
    else if (IncomingBytes[0] != '^') {
      Serial.print (IncomingBytes[0]);
      Serial.println (" Bad First Char");
    }
    else if (IncomingBytes[5] != '$') { // same as \n
      Serial.print (IncomingBytes[6],DEC);
      Serial.println (" Bad Last Char");
    }
    else  {
      Serial.println (OutgoingBytes);
    }
  }

  while (Serial.available() > 0 )
  {
    IncomingByte = Serial.read();
    Serial.print("#");
  }
  Serial.println("ControlText");
  delay(10000); // prevents overwhelming the serial port

}

Chris--A, 2015-05-19:
Do you know its receiving more characters, as it could be less, you only check for inequality `CharsRead != sizeof(IncomingBytes)`.

If your serial monitor has not been set to send a line feed, then you'll not be receiving a '\n'. You can configure this setting in the bottom right hand corner next to the baud rate selection box.

If your data takes a while to arrive, you should wait for sufficient data before reading, as `readBytesUntil()` can timeout, meaning your target is never found.

`if (Serial.available() > 0 ) {` : This will run your code looking for a string, even if only 1 character has been received.

peabo7, 2015-05-19:
The problem is here:

 char OutgoingBytes[6]= {
 '^','X','-','-','-','$'};

Strings in C and C++ are conventionally terminated by a binary 0, but you filled all six bytes of your array with message data.  Try this instead:

  char OutgoingBytes[7]= {
  '^','X','-','-','-','$', 0};

The reason it used to work is that by chance the compiler put something after the array which had a zero in it.  Since the compiler is free to put independent variables in any order it wants, the newer version happened to put the bad length message after the OutgoingBytes message.

Peter Olson

MJP007, 2015-05-19:
peabo7 appears to have cracked it.  I hadn't appreciated that null terminated strings were being used but they obviously are.  Put this one down to user error then.  Thanks very much for the rapid responses.

closed 2015-05-19 by MJP007